espressif/esp-idf #1555: Help on a first program : get started

St7ven, 2018-01-28:
Hi,

I'm using an ESP32 dev board which has a usb/uart transceiver (CP2102).

I have installed all eclipse setup, and load some program successfully.

When I supervise the chip with the terminal debugger, it show me that the MCU is restarting, again and again :  

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d0862  PS      : 0x00060333  A0      : 0x800d0c26  A1      : 0x3ffb4bd0  
A2      : 0x3ffb5e28  A3      : 0x3ffb4bf0  A4      : 0x3ffb5f64  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0xff000000  A8      : 0xffffffff  A9      : 0x00000001  
A10     : 0x3ffb4cf4  A11     : 0x00060323  A12     : 0x00060323  A13     : 0x00000001  
A14     : 0x00060320  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0862:0x3ffb4bd0 0x400d0c23:0x3ffb4bf0 0x400d0c53:0x3ffb4c20 0x400d178d:0x3ffb4c40 0x4008481e:0x3ffb4c60

Rebooting...
```
The code is quite simple : 

```
/* Blink Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#include "esp_log.h"

/* Can run 'make menuconfig' to choose the GPIO to blink,
   or you can edit the following line and set a number here.
*/
#define BLINK_GPIO CONFIG_BLINK_GPIO

void blink_task(void *pvParameter)
{
    /* Configure the IOMUX register for pad BLINK_GPIO (some pads are
       muxed to GPIO on reset already, but some default to other
       functions and need to be switched to GPIO. Consult the
       Technical Reference for a list of pads and their default
       functions.)
    */
    int i =0;

    gpio_pad_select_gpio(BLINK_GPIO);
    /* Set the GPIO as a push/pull output */
    gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);
    while(1) {
        /* Blink off (output low) */
        gpio_set_level(BLINK_GPIO, 0);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        /* Blink on (output high) */
        gpio_set_level(BLINK_GPIO, 1);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        ESP_LOGI("DEBUG", "Blinked %i", i);
    }
}

void app_main()
{
    xTaskCreate(blink_task, "blink_task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);

    // Start the real time scheduler.
 //   vTaskStartScheduler();

}

```

The MCU reset just after ESP_LOGI has printed Blinked 0

By the way, I did not need to have the vTaskStartScheduler(); to start the task. I don't know why...

negativekelvin, 2018-01-28:
configMINIMAL_STACK_SIZE is too small for using ESP_LOG functions using vprintf

St7ven, 2018-01-28:
It did the trick. I add 4k. 

I met so much issue to solve, I guess I did not try this one hard enough. Thank you for your help. Appreciate it.

closed 2018-01-28 by St7ven