espressif/esp-idf #1558: Exec format error. Suggests path but I think my paths are fine.

KapnK, 2018-01-29:
I followed the procedure at http://esp-idf.readthedocs.io/en/latest/get-started/index.html#get-started-get-esp-idf and got these errors. I deleted everything and started over and get the same thing.

`pi@raspberrypi:~/esp/hello_world $ make flash
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/bootloader_support/src/bootloader_flash.o
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
/home/pi/esp/esp-idf/make/component_wrapper.mk:273: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 2
/home/pi/esp/esp-idf/make/project.mk:450: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/home/pi/esp/esp-idf/components/bootloader/Makefile.projbuild:40: recipe for target '/home/pi/esp/hello_world/build/bootloader/bootloader.bin' failed
make: *** [/home/pi/esp/hello_world/build/bootloader/bootloader.bin] Error 2
pi@raspberrypi:~/esp/hello_world $ `

`pi@raspberrypi:~/esp/hello_world $ printenv PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games:/home/pi/esp/xtensa-esp32-elf/bin`

jcmvbkbc, 2018-01-29:
> I followed the procedure at http://esp-idf.readthedocs.io/en/latest/get-started/index.html#get-started-get-esp-idf and got these errors
> pi@raspberrypi:~/esp/hello_world $ make flash /bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error 

It looks like your build system is ARM-based Raspberry PI. But the instructions you mentioned are for x86(_64)-based PC. You'd need to rebuild the toolchain for your ARM host.

igrr, 2018-01-29:
@KapnK as @jcmvbkbc has pointed out, you are using an ARM platform, but the prebuilt toolchain packages are for x86_64 and i686. We don't provide pre-built ARM toolchain at the moment, but you can follow comments in this issue if you need to build one:
https://github.com/espressif/esp-idf/issues/589#issuecomment-301013299

KapnK, 2018-01-29:
Yes I am. I thought the 32bit x86 would work. I first attempted it a couple months ago and built from source but I couldn't get it to work so I thought I read I could just clone the 32bit and I guess not. So I should build from source according to the Linux/Debian instructions? I'm using stretch in case you've heard of any issues with that release. Thanks for the replies. It's frustrating getting stuck right at the start.

KapnK, 2018-01-30:
I fired off the build before bed, last night, according to this page utilizing the Ubuntu 16.04 package list: [http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup-scratch.html](url). This morning I was able to flash the "hello_world" example and monitor. This exercise jarred my memory and now I remember why my initial attempt didn't work. I created the toolchain ...elf/bin PATH under ~/esp/espressif-esp32-elf/bin but the build from scratch procedure puts elf/bin under ct-ng/builds/. Thanks for your help!

closed 2018-02-07 by FayeY