espressif/esp-idf #1559: Set other partition active on panic

SteveOfTheStow, 2018-01-29:
Has consideration ever been given to a facility to set the other partition to active (assuming an OTA partition table) on panic? This would help recover from dodgy software downloads where the partition boots but fails inside the app code - would be really useful in production.

i.e.
- Partition A does an OTA to partition B, and B is set as the active partition
- ESP32 restarts, Partition B boots, but something is wrong inside the app code and it faults
- Panic handler use the ESP partition API to change the active partition to A
- Partition A, which is good, boots.

If an OTA were attempted again and B was bad, the same result would occur - a restore to A, meaning negligible chance of bricking the chip.

igrr, 2018-01-29:
Changing the active partition might be hard to do *reliably* in the panic handler, as the state of the system is not well defined at that point. We do core dumps to flash from panic handler, but these are implemented as a best-effort measure; not something that is guaranteed to work in all cases.

The option we are considering is to allow the application acknowledge a successful boot following an OTA. If the application does not acknowledge successful boot and restarts N (=parameter) times, the bootloader itself can switch the active application back to the old partition.
Acknowledgement could be implemented by extending the ota_data partition to record not only the selected app, but also its state. Restart count could be maintained by the bootloader in RTC memory.
Would such an approach satisfy your requirements?

SteveOfTheStow, 2018-01-29:
Understood about the panic handler not running within a reliable context.
The application acknowledge method should satisfy my requirements, yes! What would be the timeline for that - IDF 3.1?

forthlightning, 2018-01-30:
+1 for this, was thinking about hacking something together that unset a safety bit on boot and set the bit after full initialization, and then the startup sequence would check for that bit. Hadn't even considered falling back to the other partition. This is a great idea!

TimHughey, 2018-02-24:
+1 have also been considering implementing a similar concept for my project to detect an unstable build (e.g. frequent restarts / panics) and fall back to previous version.  It does make sense for this check to be done by the bootloader vs. application.

KonstantinKondrashov, 2019-11-14:
Hi @SteveOfTheStow, @forthlightning  @TimHughey,  @milamber-ls.
This feature is named `Rollback`, it was implemented in release/v3.3 https://github.com/espressif/esp-idf/blob/release/v3.3/components/bootloader/Kconfig.projbuild#L166. Get more information here - https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/ota.html#app-rollback .

An option in Kconfig CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE allows you to track the first boot of a new application. In this case, the application must confirm its operability by calling esp_ota_mark_app_valid_cancel_rollback() function, otherwise the application will be rolled back upon reboot. It allows you to control the operability of the application during the boot phase. Thus, **a new application has only one attempt to boot successfully**.

Alvin1Zhang, 2019-11-15:
@SteveOfTheStow Thanks for reporting, feel free to reopen, thanks.

closed 2019-11-15 by Alvin1Zhang