espressif/esp-idf #1560: esp_ble_gatts_set_attr_value complaining on zero-length values

lightoze, 2018-01-29:
When `value_len` is 0 malloc in `btc_gatts_arg_deep_copy` returns NULL and "no mem" error message is produced. Probably esp_ble_gatts_send_indicate and esp_ble_gatts_send_response are affected too.

Weijian-Espressif, 2018-02-11:
@lightoze thank you for your reminder.   by the way, value_len should be set to a number greater than zero .

lightoze, 2018-02-11:
Is empty string not a valid characteristic value in GATT? I'm not an expert, but except for above issue empty strings work fine with "nRF Connect" Android app by Nordic Semiconductor. Both for reading and writing.

Weijian-Espressif, 2018-02-11:
@lightoze OK, we will fix the bug later, thanks for your reminder.

jack0c, 2022-01-05:
version since ESP-IDF v3.3 have the fix.
release/v3.3 backport: 2c353edbc4fcee79c44830ffd43fa3bc7950eb77

closed 2022-01-05 by jack0c