espressif/esp-idf #1561: ble security error

JoonDong2, 2018-01-29:
I tested gatt_securty_server example in the esp-idf on some android version and iOS.

It works well on android nougat(7.x.x) and iOS but not on android kitkat(4.x.x) and marshmallow(6.x.x).

On kitkat, it worked up to bonding but disconnected with gatt error 0x89(related with authentication).

On marshmallow, it did not even bonding with 0x89 error.

I had a similar question before, but I was told that it was reported to the development team.

Is this a bug?? If this is a bug, are you fixing it?

Yulong-espressif, 2018-01-30:
@JoonDong2 what does the IDF version you using it?

JoonDong2, 2018-01-30:
I tested by using latest version. I re-installed esp-idf yesterday.

Weijian-Espressif, 2018-03-20:
@JoonDong2  sorry for being late, We are fixing a bug about securty, you will be notified later.

EspressifApp, 2018-03-27:
@JoonDong2 please update to the latest IDF, and try?

JoonDong2, 2018-03-27:
@EspressifApp I tried it and it worked well on Android 4.4.4(Kitkat) and 6.0.1(Marchmallow).

closed 2018-03-29 by FayeY