espressif/esp-idf #1563: PSRAM vs FLASH on non-standard pins

markwj, 2018-01-30:
In components/esp32/spiram_psram.c we have:

`// IO-pins for PSRAM. These need to be in the VDD_SIO power domain because all chips we
// currently support are 1.8V parts.
// WARNING: PSRAM shares all but the CS and CLK pins with the flash, so these defines
// hardcode the flash pins as well, making this code incompatible with either a setup
// that has the flash on non-standard pins or ESP32s with built-in flash.
#define FLASH_CLK_IO      6  //Psram clock is a delayed version of this in 40MHz mode
#define FLASH_CS_IO       11
#define PSRAM_CLK_IO      17
#define PSRAM_CS_IO       16
#define PSRAM_SPIQ_IO     7
#define PSRAM_SPID_IO     8
#define PSRAM_SPIWP_IO    10
#define PSRAM_SPIHD_IO    9`

This won't work with external flash (using FLASH_CS_IO different than the default GPIO11).

The PSRAM code should read eFuses for this. In particular, it should not hard-code the flash CS.

Note that this may not be the only issue with PSRAM vs FLASH on non-standard pins. Even manually changing FLASH_CS_IO to the correct GPIO, we still get crashes on boot.

Ritesh236, 2018-02-25:
Hi,

This issue has been resolved into current Master branch or ESP32 IDF 3.0 RC1 SDK?

Or still in open state?

Ritesh236, 2018-02-26:
Hi,

We had checked by changing Flash Default chip select IO with External SPI Flash chip select and both external SPI Flash with PSRAM and internal flash memory are working fine without any issue.

So, when this changes will be updated into ESP32 IDF on GitHub?

panfeng-espressif, 2018-02-28:
Hello markwj,
How about using the latest IDF to solve this issue? Are you still facing with this issue?

Ritesh236, 2018-02-28:
We had updated SPI Flash CS Pin Configuration with External Flash memory and able to resolve that issue but need to make it configurable like into make menuconfig selection or something like that

markwj, 2018-03-02:
> How about using the latest IDF to solve this issue? Are you still facing with this issue?

I checked, but can't see any changes to components/esp32/spiram_psram.c since Dec 19 2017. The same #define statements and the same hard-coded values used in static void IRAM_ATTR psram_gpio_config.

Issue remains in master branch.

Ritesh236, 2018-03-03:
Yes. Issue remains same into ESP32 IDF 3.0 RC1 and latest master IDF.

So, They need to fix it by reading EFUSE parameters accordingly as per configurations done by USER.

panfeng-espressif, 2018-03-07:
Actually these pin cannot be changed unless you change software along with hardware. The FLASH_CS_IO is connected with GPIO11 in hardware.
Have you try to disconnect the connection of flash cs and gpio11? Maybe that will work

negativekelvin, 2018-03-07:
@panfeng-espressif in *some* hardware, but #defines should be removed and read pin assignments from efuse is better

panfeng-espressif, 2018-03-07:
Ok, I will show your suggestion to the developer.
At the same time, efuse for esp32 is not very big so that we have to save the space as much as we can. You know that efuse is a one-time-memory. When it is burned, the value cannot be changed again.

negativekelvin, 2018-03-07:
@panfeng-espressif that section of efuse is already used for spi so it doesn't change anything.

costaud, 2018-03-08:
Hello,  something to be confirmed:
@markwj @Ritesh236
1. Are you using an external flash, sharing data and clock wires with the original spi flash, only the CS wire is different.
2. Meanwhile you are running flash cache on the external flash, right?

costaud, 2018-03-08:
@negativekelvin 
Are you making your own module with custom IOs for SPI flash?

Something need to be noted that:
1. If you are using a 1.8V flash, the number of IOs within 1.8v power domain is very limited. Make sure you are using a correct one.
2. The timing of psram and flash is critical to the whole system. For now, we can guarantee that the default Flash and Psram pins can work perfectly. It is recommended to keep the default IO settings.

negativekelvin, 2018-03-08:
@costaud yes I understand.  For example D2WD has 1.8v flash but swaps some pins.

Also does flash cache and sram cache devices have to be on same SPI port?

Ritesh236, 2018-03-10:
Hi,

Yes. We had used ESP32-WROVER-I module with external flash memory as 16 MB. So, We have connected all pins as same with internal flash memory except CS GPIO.

So, We require more flash memory as per project requirement and for that we don't have much MOQ to create custom module.

Right now, We had updated all SPI pins settings as same except CS and updated CS according to GPIO taken. Then updated spiram_psram.c file for that custom CS and then we are able make external memory as bootable flash memory and also able to access internal flash memory + PSRAM as well without any issue.

Let me know if need anything else from my side.

costaud, 2018-03-12:
@Ritesh236 OK, I see. We will try to support those modes for PSram and Flash. But this is a feature related to the whole system, we need some more time to figure out which is the best approach, as well as to make full test for that.

costaud, 2018-03-12:
@negativekelvin , yes, flash and sram cache have to be on same SPI port. Actually, only SPI0, which is system private, has access to cache.

rahulbpatel8, 2019-04-16:
Hi ALL,

We have used external flash by efuse and using it as main flash with wrover module and facing the issue while using PSRAM as FLASH_CS_IO will need to be new efuse IO.

	**const uint32_t spiconfig = ets_efuse_get_spiconfig();**
	**if (spiconfig == EFUSE_SPICONFIG_SPI_DEFAULTS) {**
		gpio_matrix_out(FLASH_CS_IO, SPICS0_OUT_IDX, 0, 0);
		gpio_matrix_out(PSRAM_SPIQ_IO, SPIQ_OUT_IDX, 0, 0);
		gpio_matrix_in(PSRAM_SPIQ_IO, SPIQ_IN_IDX, 0);
		gpio_matrix_out(PSRAM_SPID_IO, SPID_OUT_IDX, 0, 0);
		gpio_matrix_in(PSRAM_SPID_IO, SPID_IN_IDX, 0);
		gpio_matrix_out(PSRAM_SPIWP_IO, SPIWP_OUT_IDX, 0, 0);
		gpio_matrix_in(PSRAM_SPIWP_IO, SPIWP_IN_IDX, 0);
		gpio_matrix_out(PSRAM_SPIHD_IO, SPIHD_OUT_IDX, 0, 0);
		gpio_matrix_in(PSRAM_SPIHD_IO, SPIHD_IN_IDX, 0);
	}

So using ets_efuse_get_spiconfig() before configuring GPIOs is proper solution.?  We are using v3.1-beta1-216-gd91c42517 (v3.1-beta1 branch). I am not seeing any changes on latest master either regarding this. Let me know if I am wrong.

> @Ritesh236 OK, I see. We will try to support those modes for PSram and Flash. But this is a feature related to the whole system, we need some more time to figure out which is the best approach, as well as to make full test for that.

So can we consider above as a solution. Will there be any update regarding this in near future?
Suggestions will be appreciated.

Thanks.

Ritesh236, 2019-04-16:
@panfeng-espressif & @costaud

Hi,

Would you please check last response with concern raised by Rahul which is working with me? Please provide your suggestions and inputs for same.

Ritesh236, 2019-04-18:
Respected ESP32 SDK Developers,

Would you please check comments provided by one of developer and provide your feedback for same after doing analysis from your end?

costaud, 2019-04-26:
Sorry for late reply, actually we have started a Merge Request in our internal Repo. I think it will be merged soon.

Generally, what Rahul suggested is correct. We will also read the pin configuration in EFUSE first.

allanchen24, 2019-06-05:
Hi All,
We have supported the function that auto configure psram io based on efuse, please have a try with the lastest IDF.

Alvin1Zhang, 2019-06-14:
@Ritesh236 @markwj @rahulbpatel8 Thanks for your patience and sorry for getting back to you so late. Would you help have a try with the latest IDF and help share if any updates? Thanks.

Alvin1Zhang, 2019-10-12:
@markwj @rahulbpatel8 @Ritesh236 Thanks for your patience and sorry for getting back to you so late. Would you help have a try with the latest IDF and help share if any updates? Thanks.

Ritesh236, 2019-10-13:
Hello @Alvin1Zhang ,

Sorry for late reply.

Right now, We are using ESP32 IDF 3.2 Stable Release for our product development purpose. Let me check issue and changes into that IDF and will let you know results for same.

Regards,
Ritesh Prajapati

Alvin1Zhang, 2020-05-27:
@Ritesh236 Would you please help share if any updates available? Thanks.

costaud, 2020-06-11:
Already supported on latest master branch. 
I'll close this issue, feel free to reopen if you find any problems.

closed 2020-06-11 by costaud