espressif/esp-idf #1565: How to set or get class of device of bluetooth classic

my-abousamra, 2018-01-30:
Hi,

How to set or get class of device of bluetooth classic for my device as I only find functions to process class of device to get major/minor device

Regards

Weijian-Espressif, 2018-02-11:
@blueMoodBHD Please help solve this problem.

blueMoodBHD, 2018-02-12:
@my-abousamra 
Call function  esp_bt_gap_cancel_discovery();
Then esp_bt_gap_cb_t will is called with ESP_BT_GAP_DISC_RES_EVT, in param->disc_res.prop[],if there is one that param->disc_res.prop[i].type == ESP_BT_GAP_DEV_PROP_COD, then param->disc_res.prop[i].val is the class of device.

my-abousamra, 2018-02-12:
@blueMoodBHD 
1.  Is this the way to get the current running device COD, not a remote one? I thought it is a usual function to get it.  
2.  How to set it to my device?

blueMoodBHD, 2018-03-05:
@my-abousamra We do not have the APIs to set and read cod at this moment, we will provide those APIs as soon as possible, Thanks

blueMoodBHD, 2018-03-14:
@my-abousamra Now, there is no API to set and get  the current running device COD, and we will add the APIs ASAP.

my-abousamra, 2018-03-20:
@blueMoodBHD any updates, please?

blueMoodBHD, 2018-03-21:
@my-abousamra We will try to finish this in v3.1, thanks.

my-abousamra, 2018-03-21:
@blueMoodBHD when will it be released, please?

my-abousamra, 2018-04-19:
Works fine, Thanks!

closed 2018-04-19 by my-abousamra