espressif/esp-idf #1566: Bluetooth classic SPP initiator doesn't start while WiFi starting

my-abousamra, 2018-01-30:
I tried spp initiator example and it worked well but when I integrated it with wifi it doesn't start (doesn't receive ESP_SPP_INIT_EVT event in response to esp_spp_init function call) and wifi code continues initing successfully but when stop wifi, bluetooth continues initing successfully and the event ESP_SPP_INIT_EVT comes then. any idea why and how to overcome?

Regards

my-abousamra, 2018-01-31:
[test](https://github.com/espressif/esp-idf/issues/941)

Weijian-Espressif, 2018-02-11:
@blueMoodBHD please help review the issue.

TianHao-Espressif, 2018-04-03:
@my-abousamra , for you second problem, it's due to the wifi/bluetooth coexistence function have something problem. We have a commit in latest master branch, you can try it.

my-abousamra, 2018-04-04:
@TianHao-Espressif I tried latest master commit but I have this concern
WiFi starts fine while bluedriod is enabled but it fails to continue setting up SSL as it suffers from heap run out.
It seems that bluedroid is heap hungery. I disabled it and checked free heap which was about 264KB at application starting and after enable it and check again it dropped to about 142KB which is not suitable to start SSL. According to this values, Bluedroid consumes about 122KB of available RAM which is 520KB (almost 23% of memory!)

Now I'm stuck on this point and I am afraid of not being able to start the 2 modules at the same time bluedroid and SSL.

TianHao-Espressif, 2018-04-08:
@my-abousamra, not the bluedroid consume much memory, the controller need almost 70K memory if enable dual mode. I you don't need BLE, please call `ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));` and do not enable GATT things in menuconfig. 

For memory lacking, another solution is "add PSRAM".

TianHao-Espressif, 2018-06-04:
@my-abousamra , Does your board have PSRAM? I suggest you use PSRAM when you will run wifi and classic bt while running a big application. But if your application need a little memory, you needn't to use PSRAM.

my-abousamra, 2018-06-04:
No, I don't use one

closed 2018-07-12 by FayeY