espressif/esp-idf #1571: Use of LWIP option MEMP_MEM_MALLOC makes protocol PCB configuration nonfunctional

mill1000, 2018-02-01:
In ESP-IDF v3.0-rc1, the use of LWIP option MEMP_MEM_MALLOC makes the configuration options for protocol PCBs do nothing, At least CONFIG_LWIP_MAX_RAW_PCBS and CONFIG_LWIP_MAX_UDP_PCBS are useless parameters.

zhangyanjiaoesp, 2018-03-15:
@mill1000 Can you describe your problem in detail?

mill1000, 2018-03-15:
Less of a problem, more of an observation.

In lwipopts.h, MEMP_MEM_MALLOC is defined to 1 which tells LWIP to use malloc/free instead of its internal memory pools.

As memory pools are disabled the following configuration options have no effect. They are used to set the mempool sizes, which do not exist.
- CONFIG_LWIP_MAX_RAW_PCBS
- CONFIG_LWIP_MAX_LISTENING_TCP
- CONFIG_LWIP_MAX_UDP_PCBS

These configs options still serve a purpose as they control more than just the mempool size
- CONFIG_LWIP_MAX_ACTIVE_TCP
- CONFIG_LWIP_MAX_SOCKETS

zhangyanjiaoesp, 2018-04-08:
@mill1000  
In the original LWIP code, the `CONFIG_LWIP_MAX_SOCKETS` is also used to control more than just the mempool size, so I think what you said is not a problem. Maybe you can close this issue.

closed 2018-04-08 by FayeY