espressif/esp-idf #1573: Crash in btdm_bb_isr when connecting to WiFi AP

ColinFitzgerald, 2018-02-01:
Hi all !

I'm getting a strange crash while trying to use BLE and WiFi at the same time.  The crash seems to happen when I'm connected (to iPhone) with BLE and then attempt to establish a WiFi connection to an AP.  Looks like some kind of conflict between phy_init and a BT ISR.  This may be a bug, but may also be not appropriate usage of the device??  The crash is repeatable but doesn't happen all the time.

This is part of a big project so may take some time to cut down to a simple example if needed.  Just looking for a starting point.  It seems though this shouldn't be happening.

Using Pre-release 3.0-rc1

`I (27233) WIFI: Connecting...
I (27281) BLE GATTS: ESP_GATTS_READ_EVT, Connection ID: 0, Transfer Id: 32, Handle: 77, Need Response: No
I (27853) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (27853) wifi: state: init -> auth (b0)
I (27861) wifi: state: auth -> assoc (0)
I (27869) wifi: state: assoc -> run (10)
I (27893) wifi: connected with Plantiga Office, channel 1
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x40192a3c  PS      : 0x00060034  A0      : 0x80087559  A1      : 0x3ffc05c0  
0x40192a3c: phy_enter_critical at /home/colin/sdk/esp-idf/components/esp32/./phy_init.c:50

A2      : 0x00e0a000  A3      : 0xb0000000  A4      : 0x00060023  A5      : 0x000000ff  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8008da81  A9      : 0x3ffc05a0  
A10     : 0x00000001  A11     : 0x3ffc05c0  A12     : 0x3ffc05c8  A13     : 0x0000a9d4  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
Core 0 was running in ISR context:
EPC1    : 0x40062221  EPC2    : 0x4000bff0  EPC3    : 0x00000000  EPC4    : 0x40192a3c
0x40192a3c: phy_enter_critical at /home/colin/sdk/esp-idf/components/esp32/./phy_init.c:50

Backtrace: 0x40192a3c:0x3ffc05c0 0x40087556:0x3ffc05f0 0x40082b81:0x3ffc0610 0x4006221e:0x00000000
0x40192a3c: phy_enter_critical at /home/colin/sdk/esp-idf/components/esp32/./phy_init.c:50

0x40087556: btdm_bb_isr at intc.c:?

0x40082b81: _xt_lowint1 at /home/colin/sdk/esp-idf/components/freertos/./xtensa_vectors.S:1105

CPU halted.
`

Elektrik1, 2018-02-01:
Same here, no WiFi, only classic BT (a2dp_sink example), also errors on pair/connect:

	E (81812) BT: btm_sec_conn_req
	E (81822) BT: btm_sec_connected
	Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region
	 accessed)
	Core 0 register dump:
	PC      : 0x40084949  PS      : 0x00060034  A0      : 0x80004160  A1      : 0x3f
	fc0530
	0x40084949: ram_chip_i2c_readReg at /home/cff/gittree/chip7.1_phy/chip_7.1/board
	_code/app_test/pp/phy/phy_chip_v7_ana.c:1802

	A2      : 0x00000062  A3      : 0x00000001  A4      : 0x00000005  A5      : 0x3f
	fdb240
	A6      : 0x3ffae270  A7      : 0x00060123  A8      : 0xbad00bad  A9      : 0x3f
	404aa8
	A10     : 0x00060023  A11     : 0x3ffdb220  A12     : 0xffffffff  A13     : 0x3f
	fdb210
	A14     : 0x3ffae270  A15     : 0x3ffdb240  SAR     : 0x00000015  EXCCAUSE: 0x00
	000007
	EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00
	000000
	Core 0 was running in ISR context:
	EPC1    : 0x40062230  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40
	084949
	0x40084949: ram_chip_i2c_readReg at /home/cff/gittree/chip7.1_phy/chip_7.1/board
	_code/app_test/pp/phy/phy_chip_v7_ana.c:1802

	Backtrace: 0x40084949:0x3ffc0530 0x4000415d:0x3ffc0550 0x400041da:0x3ffc0570 0x4
	0084a45:0x3ffc0590 0x40084d64:0x3ffc05b0 0x40087cfe:0x3ffc05e0 0x40081e16:0x3ffc
	0600 0x4006222d:0x00000000
	0x40084949: ram_chip_i2c_readReg at /home/cff/gittree/chip7.1_phy/chip_7.1/board
	_code/app_test/pp/phy/phy_chip_v7_ana.c:1802

	0x40084a45: pll_correct_dcap at /home/cff/gittree/chip7.1_phy/chip_7.1/board_cod
	e/app_test/pp/phy/phy_chip_v7_ana.c:1802

	0x40084d64: bt_track_pll_cap at /home/cff/gittree/chip7.1_phy/chip_7.1/board_cod
	e/app_test/pp/phy/phy_chip_v7_ana.c:1802

	0x40087cfe: btdm_bb_isr at intc.c:?

	0x40081e16: _xt_lowint1 at C:/msys32/home/esp/esp-idf/components/freertos/xtensa
	_vectors.S:1105

ColinFitzgerald, 2018-02-01:
I've updated to the very latest commits and now see this crash happening exactly as Elektrik1, in phy_chip_v7_ana.c

igrr, 2018-02-02:
Thanks for the reports folks! The original issue reported by @ColinFitzgerald was fixed in https://github.com/espressif/esp-idf/commit/95af1c22b5c46170aa8508f18a5b8d0a0d2b19ff, but there was an unrelated change in https://github.com/espressif/esp-idf/commit/2242bf9b37efa900c6e2e10ab89e9c2afc4e2701#diff-d7a5610415927bcc3e2c66fe5f0680eb which moved libphy.a's .rodata from DRAM into flash. This caused the second version of the issue which you are observing in the latest git version. 

We'll have the fix shortly; in the meantime, you can revert the deletion of `*libphy.a:(.rodata .rodata.*)` line in components/esp32/ld/esp32.common.ld (see diff link above).

closed 2018-02-02 by igrr

ColinFitzgerald, 2018-02-02:
Awesome! Thank-you for the super quick response! Loving the ESP32 and it's ecosystem!! <3