espressif/esp-idf #1576: Docs: "Heap Tracing To Find Heap Corruption"

PerMalmberg, 2018-02-03:
The section [Heap Tracing To Find Heap Corruption](https://esp-idf.readthedocs.io/en/latest/api-reference/system/heap_debug.html?highlight=corrupt#heap-tracing-to-find-heap-corruption) does not contain any actual information, seems a large section of it has been taken out, or never written.

negativekelvin, 2018-02-03:
https://github.com/espressif/esp-idf/commit/04188d8ec74c1f5c58e4582c2ac0c45717a40f33#diff-89a297e22b11c41cc9544fd4d88b74ba

projectgus, 2018-02-05:
Thanks for reporting this. It looks like my brain cut itself off in mind sentence and I never finished those thoughts. Probably something like:

```
Using heap tracing in this way is very similar to memory leak detection as described above. For memory which is allocated and not freed while tracing is running, the entry in a HEAP_TRACE_ALL record and a HEAP_TRACE_LEAKS record will be the same. However, the HEAP_TRACE_ALL trace will also contain memory which was freed during the trace run (so you may be able to see the same region reused multiple times.
```

The second cut-off sentence I think is probably just editing cruft that I did not remove before comitting.

At the moment there is some room to add better tooling for correlating heap trace entries with the address at which corruption occured. At the moment you have to do this manually. There is a request open in #1316 to make it easier to programatically do this, but it's not implemented yet.

projectgus, 2018-12-04:
Edited this section in c9cbc6b30cb9afb012d1f592feb8b33a3db1b8a7, forgot to update the issue until now (oops!)

closed 2018-12-04 by projectgus