espressif/esp-idf #1578: disable system printf

xarrange, 2018-02-03:
is it possible to get rid of all system messages (boot, wifi, etc...) in UART but leave the option to use printf?

negativekelvin, 2018-02-03:
https://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html#_CPPv219esp_log_set_vprintf14vprintf_like_t

xarrange, 2018-02-03:
that works for wifi messages, but bootloader still prints its stuff

negativekelvin, 2018-02-04:
https://github.com/espressif/esp-idf/blob/master/components/bootloader/Kconfig.projbuild#L2

xarrange, 2018-02-04:
nope. still prints:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
....

negativekelvin, 2018-02-04:
That's ROM bootloader, you have to pull MTDO low

xarrange, 2018-02-04:
Wow! that helped, thanks!

closed 2018-02-04 by xarrange