espressif/esp-idf #1581: vApplicationIdleHook activated but not called

St7ven, 2018-02-04:
Hi,

I'm getting this annoying message on the uart/usb port of my ESP32 every 5sec (as the wdt is configured):
```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
```
So, I tried to called this function : vApplicationIdleHook() to feed the watchdog during idle time (esp_task_wdt_reset()). I add a printf function to the idle hook as well in order to supervise this function.

I have enabled the legacy for "idle hook" in the make menuconfig.

Nevertheless, the vApplicationIdleHook() has never been called.

What do I missed ?

[main.zip](https://github.com/espressif/esp-idf/files/1693071/main.zip)

negativekelvin, 2018-02-04:
The idle task has to be the one to call those functions meaning your main task has to not hog the CPU by sleeping or waiting for a semaphore, etc.

St7ven, 2018-02-04:
Thanks for the reply !
In my current project, the task does not hog the CPU (I assume) because they call the vTaskDelay() function for a period of 1sec. Please, see the previous file "main.zip".

negativekelvin, 2018-02-04:
The main task has while loop at the end with nothing inside.  Use gist instead of attaching files.

St7ven, 2018-02-04:
Ok, I see now what you mean. 

Thank you for taking time. I'm still learning a lot of this rtos and you have been very helpfull.

closed 2018-02-04 by St7ven