espressif/esp-idf #1583: Configurable binary name

SteveOfTheStow, 2018-02-05:
Looking at project.mk, the name of the project binary appears to be hardcoded:

`APP_ELF:=$(BUILD_DIR_BASE)/$(PROJECT_NAME).elf
APP_MAP:=$(APP_ELF:.elf=.map)
APP_BIN:=$(APP_ELF:.elf=.bin)`

I'd like to append a version to my binary filename, which is sourced from a file in my project, but no obvious way to do it.
I can submit a pull request to enable it to be overridden but wanted to check I haven't missed some already implemented or planned functionality?

igrr, 2018-02-11:
The easiest option would be to set `PROJECT_NAME` variable in your project makefile to include the version. Would that work for your case?

igrr, 2018-03-20:
@SteveOfTheStow Please feel free to reopen if the suggested approach does not work for you.

closed 2018-03-20 by igrr