espressif/esp-idf #1584: Refresh of webpage from controller that has a running websocket causes crash of controller when using WiFi, but not Ethernet

agmuntianu, 2018-02-05:
Hi,

I have started using the the HTTPD component that @mjahamel made a pull request for (https://github.com/espressif/esp-idf/pull/1371 , I use it as a local component in my project ).
I have a websocket that starts streaming data back and forth from the controller to the webpage.
The webpage loads fine, but if I refresh the page at ~1-2 seconds interval I get the controller to eventually crash.
This crash only happens if I use the WiFi ( router is ~1 m away from esp32 ) . I also have Ethernet implemented and when using it , I do not experience any crashes.

This is the stacktrace I get most of the time:
```
CORRUPT HEAP: multi_heap.c:370 detected at 0x3ffcccd8
abort() was called at PC 0x40088803 on core 1
0x40088803: multi_heap_assert at /home/munt/esp/esp-idf/components/heap/./multi_heap_platform.h:55
 (inlined by) multi_heap_malloc_impl at /home/munt/esp/esp-idf/components/heap/./multi_heap.c:370

Backtrace: 0x40088fd8:0x3ffc3d10 0x400890d7:0x3ffc3d30 0x40088803:0x3ffc3d50 0x40082e00:0x3ffc3d70 0x40082e31:0x3ffc3d90 0x400832d5:0x3ffc3db0 0x4000beaf:0x3ffc3dd0 0x400f6a86:0x3ffc3df0 0x401042a5:0x3ffc3e10 0x40104b93:0x3ffc3e30 0x40104cc8:0x3ffc3e60 0x40100dd5:0x3ffc3e80 0x400f727a:0x3ffc3ea0 0x400f668e:0x3ffc3ec0 0x400fc55d:0x3ffc3ee0
0x40088fd8: invoke_abort at /home/munt/esp/esp-idf/components/esp32/./panic.c:572
0x400890d7: abort at /home/munt/esp/esp-idf/components/esp32/./panic.c:572
0x40088803: multi_heap_assert at /home/munt/esp/esp-idf/components/heap/./multi_heap_platform.h:55
 (inlined by) multi_heap_malloc_impl at /home/munt/esp/esp-idf/components/heap/./multi_heap.c:370
0x40082e00: heap_caps_malloc at /home/munt/esp/esp-idf/components/heap/./heap_caps.c:401
0x40082e31: heap_caps_malloc_default at /home/munt/esp/esp-idf/components/heap/./heap_caps.c:401
0x400832d5: _malloc_r at /home/munt/esp/esp-idf/components/newlib/./syscalls.c:37
0x400f6a86: pbuf_alloc at /home/munt/esp/esp-idf/components/lwip/core/pbuf.c:1247
0x401042a5: tcp_output_alloc_header at /home/munt/esp/esp-idf/components/lwip/core/tcp_out.c:113
0x40104b93: tcp_send_empty_ack at /home/munt/esp/esp-idf/components/lwip/core/tcp_out.c:914
0x40104cc8: tcp_output at /home/munt/esp/esp-idf/components/lwip/core/tcp_out.c:1008
0x40100dd5: tcp_input at /home/munt/esp/esp-idf/components/lwip/core/tcp_in.c:478
0x400f727a: ip4_input at /home/munt/esp/esp-idf/components/lwip/core/ipv4/ip4.c:711
0x400f668e: ethernet_input at /home/munt/esp/esp-idf/components/lwip/netif/ethernet.c:176
0x400fc55d: tcpip_thread at /home/munt/esp/esp-idf/components/lwip/api/tcpip.c:474
```

I tried to track down the memory leak by elevating the heap debugging to Light and Comprehensive and by adding heap_caps_check_integrity_all() in httpd() and my websocket callback, but it never detects any issue in my code, the crashes are still in LWIP component.

And this is the crash I usually get with elevated heap debugging:
```
CORRUPT HEAP: Bad tail at 0x3ffbc59c. Expected 0xbaad5678 got 0xbaad5600
CORRUPT HEAP: Bad tail at 0x3ffbc59c. Expected 0xbaad5678 got 0xbaad5600
CORRUPT HEAP: Bad tail at 0x3ffbc59c. Expected 0xbaad5678 got 0xbaad5600
CORRUPT HEAP: Bad tail at 0x3ffbc59c. Expected 0xbaad5678 got 0xbaad5600
CORRUPT HEAP: Bad tail at 0x3ffbc59c. Expected 0xbaad5678 got 0xbaad5600
CORRUPT HEAP: Bad tail at 0x3ffbc59c. Expected 0xbaad5678 got 0xbaad5600
CORRUPT HEAP: Bad tail at 0x3ffbc59c. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/home/munt/esp/esp-idf/components/heap/./multi_heap_poisoning.c", line 199, function: multi_heap_free
abort() was called at PC 0x4011017b on core 0
0x4011017b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4008929c:0x3ffc5fa0 0x4008939b:0x3ffc5fc0 0x4011017b:0x3ffc5fe0 0x40088f81:0x3ffc6010 0x40082f1a:0x3ffc6030 0x4008330d:0x3ffc6050 0x4000bec7:0x3ffc6070 0x400f12d5:0x3ffc6090 0x400e9e17:0x3ffc60b0 0x400ec051:0x3ffc60e0
0x4008929c: invoke_abort at /home/munt/esp/esp-idf/components/esp32/./panic.c:572
0x4008939b: abort at /home/munt/esp/esp-idf/components/esp32/./panic.c:572
0x4011017b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x40088f81: multi_heap_free at /home/munt/esp/esp-idf/components/heap/./multi_heap_poisoning.c:313
0x40082f1a: heap_caps_free at /home/munt/esp/esp-idf/components/heap/./heap_caps.c:401
0x4008330d: _free_r at /home/munt/esp/esp-idf/components/newlib/./syscalls.c:42
0x400f12d5: esf_buf_recycle at ??:?
0x400e9e17: ppProcTxDone at ??:?
0x400ec051: ppTask at ??:?
```

I also tried to add a delay before I start the websocket , but the crashes still happen.

BR. Andrei.

agmuntianu, 2018-02-05:
I use IDF release/v3.0

liuzfesp, 2018-02-22:
Hi @agmuntianu could you let me know the commit ID of IDF you used?

agmuntianu, 2018-02-26:
8856cc055a7d2b0e2a3dd12d1741f5b9b8ea6507

pushtoopen, 2018-02-28:
Having the same "0x4011017b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
" failure, mine is from trying to pass in a char string pointer into the netconn_write thats been read from a spiffs file structure. Not using 3.0

liuzfesp, 2018-03-05:
@zhangyanjiao please help...

agmuntianu, 2018-03-13:
Hi,
I have found the root cause of my crash, it was in the httpd module I used ,
In function `http_recv(void *arg, struct tcp_pcb *pcb, struct pbuf *p, err_t err)`, the 'hs' retries counter is being set to 0 after a possible freeing of the 'http_state' structure, this happens on refresh of webpage . I could not reproduce it on ethernet due to lower latency with it.
So basically `hs->retries = 0;` needs to be executed before:
```
        if (err == ERR_CLSD) {
            LWIP_DEBUGF(HTTPD_DEBUG, ("received close connection %p %p\n",pcb,hs));
            http_close_conn(pcb, hs);
        }
```
BR. Andrei.

closed 2018-03-13 by agmuntianu