espressif/esp-idf #1586: ESP-IDF - error in file find while adding component to project.

zafersn, 2018-02-07:
hi.  I am working on Eclipse and am using esp-idf. I am working on Eclipse and am using esp-idf. I want to communicate with the cloud using IBM-watson-iot libraries. However, although I struggled a lot, I still could not include this library in the project. The operations I have done to include the library in the project and the errors I get are as follows. What would you suggest me to solve the problem?

I compile the project on **windows OS**.

1.  I downloaded the Embedded C Client Library from repository
2.  I extracted the files and ran the setup.sh file.
3.  I combined the src file and the lib files.
4.  then I moved this lib file to the components folder in the project file I created in Eclipse.
5.  the new project structure is shown in the picture below.
[image]
6.  I created one component.mk file under the lib folder and the contents are as follows:

```
COMPONENT_ADD_INCLUDEDIRS := mbedtls
CFLAGS += -Wno-error=implicit-function-declaration -Wno-error=format= -DHAVE_CONFIG_H
```
 

7. and when I try to compile the project I get the error "components / lib / mbedtls / iotf_network_tls_wrapper.h: 27: 24: fatal error: sys / select.h: No such file or directory".

igrr, 2018-02-14:
The issue may not be related to project structure. ESP-IDF does not provide `sys/select.h`.

`select` is currently declared in `sys/socket.h` (which is the same as `lwip/sockets.h`).

igrr, 2019-04-10:
IDF now provides `sys/select.h` and `select` implementation which works both for files and sockets. Related example: https://github.com/espressif/esp-idf/tree/master/examples/system/select.

closed 2019-04-10 by igrr