espressif/esp-idf #1588: Crash in vPortCPUInitializeMutex

bernd-edlinger, 2018-02-07:
I use spi ram extension.

I saw a crash in e676676:

```
/home/ed/esp/esp-idf/components/freertos/./port.c:311 (vPortCPUInitializeMutex)- assert failed!
abort() was called at PC 0x400861af on core 0
0x400861af: vPortCPUInitializeMutex at /home/ed/esp/esp-idf/components/freertos/./port.c:386

Backtrace: 0x40088677:0x3ffd1fd0 0x400887cf:0x3ffd1ff0 0x400861af:0x3ffd2010 0x400859cb:0x3ffd2030 0x40085a82:0x3ffd2050 0x40085c3c:0x3ffd2070 0x40083eb6:0x3ffd2090 0x40083fbf:0x3ffd20b0 0x400954e0:0x3ffd20d0 0x400d3d94:0x3ffd20f0 0x400d3e54:0x3ffd2120 0x40157ecc:0x3ffd2140 0x4015d837:0x3ffd2160 0x40112576:0x3ffd2190 0x4010abb2:0x3ffd22c0 0x4010b951:0x3ffd23c0 0x4010bbce:0x3ffd2400 0x4010ef5f:0x3ffd2460 0x4010f43f:0x3ffd2480 0x4010e7c7:0x3ffd24b0 0x40115d52:0x3ffd24d0 0x40116062:0x3ffd24f0 0x401161aa:0x3ffd2520 0x40112e51:0x3ffd2550 0x400f4a4a:0x3ffd2570 0x400ee062:0x3ffd2590 0x400d39f3:0x3ffd25b0 0x400d3ce3:0x3ffd25d0 0x400d1998:0x3ffd26c0
0x40088677: invoke_abort at /home/ed/esp/esp-idf/components/esp32/./panic.c:648

0x400887cf: abort at /home/ed/esp/esp-idf/components/esp32/./panic.c:648

0x400861af: vPortCPUInitializeMutex at /home/ed/esp/esp-idf/components/freertos/./port.c:386

0x400859cb: xQueueGenericReset at /home/ed/esp/esp-idf/components/freertos/./queue.c:2037

0x40085a82: prvInitialiseNewQueue at /home/ed/esp/esp-idf/components/freertos/./queue.c:2037
 (inlined by) xQueueGenericCreate at /home/ed/esp/esp-idf/components/freertos/./queue.c:408

0x40085c3c: xQueueCreateMutex at /home/ed/esp/esp-idf/components/freertos/./queue.c:2037

0x40083eb6: lock_init_generic at /home/ed/esp/esp-idf/components/newlib/./locks.c:79

0x40083fbf: _lock_init_recursive at /home/ed/esp/esp-idf/components/newlib/./locks.c:95

0x400954e0: __sfp at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/findfp.c:145

0x400d3d94: _fopen_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fopen.c:138

0x400d3e54: fopen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fopen.c:181

0x40157ecc: mbedtls_pk_load_file at /home/ed/esp/esp-idf/components/mbedtls/library/pkparse.c:81

0x4015d837: mbedtls_x509_crt_parse_file at /home/ed/esp/esp-idf/components/mbedtls/library/x509_crt.c:1093
```

bernd-edlinger, 2018-02-07:
I believe this happens because the system is out of low memory when
my application tries to open a fils, with needs a new mutex.
The port uses pvPortMalloc to allocate the mutex.
Because this just calls malloc it may return external memory.
But vPortCPUInitializeMutex has the following check:
`configASSERT(esp_ptr_internal((const void *) mux));`

mahavirj, 2018-02-07:
Just for information, this looks similar to already reported issue here,
https://github.com/espressif/esp-idf/issues/1492

As mentioned above, fix for this is in progress, and should get integrated in IDF soon.

bernd-edlinger, 2018-02-09:
Fixed with 16de6bf.
Thanks!

closed 2018-02-09 by bernd-edlinger