espressif/esp-idf #1590: Wifi disconnect after 17f01eb

bernd-edlinger, 2018-02-07:
This happens at about the same test sequence when #1588 used to happen,
but beginning with commit 17f01eb the symptom changed to a wifi connectivity loss:

```
I (650152) wifi: bcn_timout,ap_probe_send_start
I (652652) wifi: ap_probe_send over, resett wifi status to disassoc
I (652652) wifi: state: run -> init (1)
I (652652) wifi: pm stop, total sleep time: 0/649197264

I (652652) wifi: n:2 0, o:2 1, ap:255 255, sta:2 1, prof:1
I (652662) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (655082) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (657492) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (659902) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (662322) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (664732) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (664852) wifi: n:2 1, o:2 0, ap:255 255, sta:2 1, prof:1
I (664852) wifi: state: init -> auth (b0)
I (664862) wifi: state: auth -> assoc (0)
I (664862) wifi: state: assoc -> run (10)
I (664862) wifi: connected with TP-LINK_FF3107, channel 2
I (664872) event: sta ip: 192.168.1.100, mask: 255.255.255.0, gw: 192.168.1.1
I (664872) OPC: SYSTEM_EVENT_STA_GOT_IP
I (664882) OPC: got ip:192.168.1.100

I (667862) wifi: pm start, type:0

I (671752) wifi: bcn_timout,ap_probe_send_start
I (674252) wifi: ap_probe_send over, resett wifi status to disassoc
I (674252) wifi: state: run -> init (1)
I (674252) wifi: pm stop, total sleep time: 0/6392380

I (674252) wifi: n:2 0, o:2 1, ap:255 255, sta:2 1, prof:1
I (674262) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (676682) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (679092) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (681512) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (683922) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (686332) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (688752) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (691162) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (693572) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (695992) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (698402) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (700812) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (703222) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (705642) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (708052) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (710462) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (712882) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (715292) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (717702) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (720112) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (722532) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (724942) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (727352) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (727482) wifi: n:2 1, o:2 0, ap:255 255, sta:2 1, prof:1
I (727482) wifi: state: init -> auth (b0)
I (727482) wifi: state: auth -> assoc (0)
I (727482) wifi: state: assoc -> run (10)
I (727492) wifi: connected with TP-LINK_FF3107, channel 2
I (727492) event: sta ip: 192.168.1.100, mask: 255.255.255.0, gw: 192.168.1.1
I (727492) OPC: SYSTEM_EVENT_STA_GOT_IP
I (727502) OPC: got ip:192.168.1.100

I (730492) wifi: pm start, type:0

I (749282) wifi: bcn_timout,ap_probe_send_start
I (751782) wifi: ap_probe_send over, resett wifi status to disassoc
I (751782) wifi: state: run -> init (1)
I (751782) wifi: pm stop, total sleep time: 0/21294281

I (751782) wifi: n:2 0, o:2 1, ap:255 255, sta:2 1, prof:1
I (751792) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (754212) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (756622) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (759032) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (761442) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (763862) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (766272) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (768682) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (771102) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (772472) event: station ip lost
I (773512) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (775922) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (778332) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (780752) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (783162) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (785572) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (787992) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (790402) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (792812) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (795222) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (797642) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (800052) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (802462) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (804882) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (807292) OPC: SYSTEM_EVENT_STA_DISCONNECTED
I (807412) wifi: n:2 1, o:2 0, ap:255 255, sta:2 1, prof:1
I (807412) wifi: state: init -> auth (b0)
I (807422) wifi: state: auth -> assoc (0)
I (807422) wifi: state: assoc -> run (10)
I (807432) wifi: connected with TP-LINK_FF3107, channel 2
I (807432) event: sta ip: 192.168.1.100, mask: 255.255.255.0, gw: 192.168.1.1
I (807432) OPC: SYSTEM_EVENT_STA_GOT_IP
I (807432) OPC: got ip:192.168.1.100

I (810432) wifi: pm start, type:0
```

The Messages with OPC are from my code:
static const char *TAG = "OPC";

```
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_START");
        ESP_ERROR_CHECK(esp_wifi_connect());
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_GOT_IP");
        ESP_LOGI(TAG, "got ip:%s\n",
                ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_DISCONNECTED");
        ESP_ERROR_CHECK(esp_wifi_connect());
        break;
    default:
        break;
    }
    return ESP_OK;
}
...
ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

```

bernd-edlinger, 2018-02-07:
My Setup uses spi ram, and a non-encrypted WiFi AP for simplicity.
This seems to happen at about the same time when previously #1588 used to happen.

bernd-edlinger, 2018-02-09:
Hi @jack0c,

I just noticed that your commit 3756e0c from yesterday appears to have fixed the
WiFi disconnects that I had observed previously.
That means, up to 35de8d7 this was still reproducible.

From the commit message it is not clear if this expected:
"esp32: fix esp_wifi_deinit causes assert"

I did not programmatically shut down the WiFi connection at all.
Just start it, and use if for a while.

jack0c, 2018-02-11:
@bernd-edlinger Thank you for updating. @TimXia please follow to find the root cause after Chinese New Year.

liuzfesp, 2018-02-22:
Hi @bernd-edlinger  what do you mean by saying "Just start it, and use if for a while"? Do you call esp_wifi_deinit in your application?

TimXia, 2018-02-22:
@bernd-edlinger How often does the issue happen? Will it happen if your device connects to other routers?

bernd-edlinger, 2018-02-22:
> Do you call esp_wifi_deinit in your application?

No, never.

This is the initalization code I call only once on startup:
```
/*init wifi as sta and set scan method*/
static void wifi_init(void)
{
    tcpip_adapter_init();
    ESP_ERROR_CHECK(tcpip_adapter_dhcpc_stop(ESP_IF_WIFI_STA));
    tcpip_adapter_ip_info_t ip_config; 
    IP4_ADDR (&ip_config.ip, 192, 168, 1, 100);
    IP4_ADDR (&ip_config.netmask, 255, 255, 255, 0);
    IP4_ADDR (&ip_config.gw, 192, 168, 1, 1);
    ESP_ERROR_CHECK(tcpip_adapter_set_ip_info(ESP_IF_WIFI_STA, &ip_config));
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = CONFIG_WIFI_SSID,
            .password = CONFIG_WIFI_PASSWORD,
            .scan_method = WIFI_FAST_SCAN,
            .sort_method = WIFI_CONNECT_AP_BY_SIGNAL,
            .threshold.rssi = -127,
            .threshold.authmode = WIFI_AUTH_OPEN,
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());
}

```

Except that I also initiallize NVS storage for calibration data,
intialize the fat file system, and write some data to it,
set the system time, and use mbed crypto functions.
Enable H/W support for mbed (RSA, SHA, AES)
Set the CPU speed to 240 MHz, Compile with -Os.

My application just initializes and then it uses TCP for OPC/UA.
The failure happens right in the middle of my test sequence,
at a time where a lot of short-lived tcp sockets enter time-wait2
state, at which point the low memory is used up, and the external
memory is about to be used for the first time.

This most interesting thing is that without the fix for #1588 in place
the WiFi connection breakdown prevents the other bug.

My application opens a file after each connection attempt,
and #1588 made the open function crash when the internal
mutex was allocated from ext memory.

So I assume some memory was allocated for WiFi from the
external memory instead of internal memory which made the
WiFi connectivity break down.

bernd-edlinger, 2018-02-22:
> How often does the issue happen? Will it happen if your device connects to other routers?

Unfortunately I have no other access points to try it out.
I used a "TP-LINK TL-WA701ND 150Mbps Wireless Lite N Access point"
Right out of the box, that means I did not change anything from the factory configuration
including the SSID, it was configured to "TP-LINK_FF3107" by the factory.

liuzfesp, 2018-03-05:
Hi @bernd-edlinger, just investigate this issue again, I think 3756e0c should be the solution for this issue. In 17f01eb a new issue is introduced, the WiFi driver allocates all frame buffer from internal memory, but it should allocate RX buffer from external memory firstly if external memory feature is enabled. Because of this, the internal memory may not enough for WiFi connection, 3756e0c fix s this issue.  Need you help to confirm that commit 3756e0c do solve your problem, otherwise we need to further investigate it.

bernd-edlinger, 2018-03-05:
No, that's fine, my tests indicated the same. I just wanted to make sure the issue was actually
solved and understood, and _not only hidden_ by an unrelated change, so it won't pop up again.

closed 2018-03-05 by bernd-edlinger