espressif/esp-idf #1595: ERROR - IDF MONITOR

lausellduane, 2018-02-08:
Please help I'm getting an error with IDF MONITOR and i don't know what is causing it. If someone can help me with this i will greatly appreciate it. 

duanes-mbp:hello_world duanelausellrosado$ make monitor
MONITOR
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

Traceback (most recent call last):
  File "/Users/duanelausellrosado/esp/esp-idf/tools/idf_monitor.py", line 622, in <module>
    main()
  File "/Users/duanelausellrosado/esp/esp-idf/tools/idf_monitor.py", line 545, in main
    monitor.main_loop()
  File "/Users/duanelausellrosado/esp/esp-idf/tools/idf_monitor.py", line 272, in main_loop
    self.handle_serial_input(data)
  File "/Users/duanelausellrosado/esp/esp-idf/tools/idf_monitor.py", line 306, in handle_serial_input
    self.console.write_bytes(b)
  File "/Users/duanelausellrosado/anaconda/lib/python3.6/site-packages/serial/tools/miniterm.py", line 63, in write_bytes
    self.byte_output.write(byte_string)
TypeError: a bytes-like object is required, not 'int'
make: *** [monitor] Error 1

projectgus, 2018-02-08:
idf_monitor (and ESP-IDF) is not currently compatible with Python 3.

If `python` is Python 3 on your system then you can set the name of the Python interpeter in "make menuconfig" -> "SDK Tools" to `python2`.

lausellduane, 2018-02-08:
Ok I made the change but when I `make flash` It gives me another error: 

duanes-mbp:hello_world duanelausellrosado$ make flash
Traceback (most recent call last):
  File "/Users/duanelausellrosado/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 34, in <module>
    import serial
ImportError: No module named serial
make[1]: *** [/Users/duanelausellrosado/esp/hello_world/build/bootloader/bootloader.bin] Error 1
make: *** [/Users/duanelausellrosado/esp/hello_world/build/bootloader/bootloader.bin] Error 2

I would need to install `serial` using Python 2?

projectgus, 2018-02-08:
That's right, you'll need to install pyserial with Python 2. The step is probably something like `sudo pip2 install pyserial`.

lausellduane, 2018-02-08:
I don't understand I used `sudo pip2 install pyserial` and it downloaded pyserial but it didn't solved the problem it still gives the same error.

duanes-mbp:hello_world duanelausellrosado$ sudo pip2 install pyserial
Password:
The directory '/Users/duanelausellrosado/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/duanelausellrosado/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting pyserial
  Downloading pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |████████████████████████████████| 194kB 1.9MB/s
Installing collected packages: pyserial
Successfully installed pyserial-3.4
duanes-mbp:hello_world duanelausellrosado$ make flash
Traceback (most recent call last):
  File "/Users/duanelausellrosado/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 34, in <module>
    import serial
ImportError: No module named serial
make[1]: *** [/Users/duanelausellrosado/esp/hello_world/build/bootloader/bootloader.bin] Error 1
make: *** [/Users/duanelausellrosado/esp/hello_world/build/bootloader/bootloader.bin] Error 2

projectgus, 2018-02-13:
Hi @lausellduane ,

There seems to be something odd with your computer's Python setup, maybe there are multiple Python installations so for some reason the one "pip2" is using is different to the one "python2" is using.

You may be able to help figure it out by running these two terminal commands:

`which python`
`which pip2`

(This will give the full paths to the two executables.)

chmorgan, 2018-03-07:
@projectgus open to a pull request to port idf_monitor.py to python3? I'm hitting the same issue this guy is seeing but your suggestion to change the default python should fix my issue.

The change occurred recently, homebrew changed from symlinking python to python2 to python3. It has been 10 years since python3 was released and there is a big push online to get people off of python2.

projectgus, 2018-03-12:
@chmorgan Sorry for the slow reply, I was on leave.

Very open to this, provided we can have Py2+3 support together. The only reason we haven't done it ourselves yet is lack of time.

There is an open PR #1284 that addresses most of the points for Py3 compatibility, but it still has some issues and seems the original author is no longer working on it. You could use this as a base, or send another PR - either approach is welcome.

projectgus, 2019-03-01:
Sorry this wasn't updated earlier. ESP-IDF now supports both Python 2 & 3.

closed 2019-03-01 by projectgus