espressif/esp-idf #1598: An interrupt handled by `gpio_isr_handler_add` crashes when called too often

chanibal, 2018-02-08:
I have a signal where I listen to and have to add my bits to it at precise moments. 

To do so, I listen to the beginning of the signal with a handler added by `gpio_isr_handler_add` (anyedge) and respond using RMT. 

In my interrupt handler, I check if my signal is OK (mostly some timing checks against the value from `xthal_get_ccount()`) and if it is, I notify my RMT-sending task with `sendToBackFromISR`.

Everything was working well, until I have found that sometimes the cables receive some interference. I managed to filter them out in software, but if there is too much of it, ESP crashes with a WDT timeout.

```
Gurru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x4008623d  PS      : 0x00060034  A0      : 0x400845a4  A1      : 0x3ffc05f0
0x4008623d: gpio_intr_service at (...)/esp-idf/components/driver/gpio.c:428

0x400845a4: _xt_lowint1 at (...)/components/freertos/xtensa_vectors.S:1105

A2      : 0x00000008  A3      : 0x00000020  A4      : 0x00000000  A5      : 0x00000007
A6      : 0x00000003  A7      : 0x00060923  A8      : 0x00000027  A9      : 0x3ff44000
A10     : 0x3ffc31b4  A11     : 0x00000000  A12     : 0x157035be  A13     : 0x3ffc05c0
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 00x4008d918  LEND    : 0x4008d946  LCOUNT  : 0xffffffff
0x4008d918: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x4008d946: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

Backtrace: 0x4008623d:0x3ffc05f0 0x400845a1:0x3ffc0610 0x40086e7a:0x00000000
0x4008623d: gpio_intr_service at (...)/esp-idf/components/driver/gpio.c:428

0x400845a1: _xt_lowint1 at (...)/esp-idf/components/freertos/xtensa_vectors.S:1105

0x40086e7a: xTaskGetSchedulerState at (...)/esp-idf/components/freertos/tasks.c:5074
```

This is probably caused by a massive amount of interrupts that happened when another device was powered down and started adding interference to the wire. Before the signals are in the hundred microsecond level, when the issue occurs it drops down to hundereds of nanoseconds.

How can I protect myself against such signals? I am able to filter them out, but what to do against crashes?   
Setting my interrupt to something like "after I'm called, disable me for 10us" would do the trick, but I can't see anything I could use like this.

costaud, 2018-02-27:
Are you using edge or level mode for GPIO interrupt? 

How long does your ISR handler take?

koobest, 2018-04-02:
Hi, @chanibal
Has this problem been solved?

chanibal, 2018-04-02:
@koobest, not yet. I have found a way to minimize this, but it still happens from time to time.

@costaud, sorry didn't notice your question earlier. 
> Are you using edge or level mode for GPIO interrupt?

Using `GPIO_INTR_ANYEDGE`. I act only on negedge, but need both negedge and posedge for clearing my timers. Possibly could do with only negedge with some changes.

> How long does your ISR handler take?

I don't currently have access to the device, so can't measure it, but from what I see from the sources, the handler shouldn't take much time.  
What the handler does:
1. Some basic math and timing checks - mostly checking for a flag and whether enough has time elapsed from the last interrupt. The time source is `xthal_get_ccount`.
2. Only if the timing test (the above) succeeds, the rest is executed. That means that with the situation that is in the issue, the timing test above will be enough to cause the crash.
3. A quick filter that fixes most my noise issues - `if(gpio_get_level(pin)) { glitches++; return; }` - without this it would crash 10x times more often. This filter comes after timing checks, so I can know that at this point that the signal is correct (a negedge after at least 12µs without glitches).
4. And only if there was enough time before last interrupt, there is a `xQueueSendToBackFromISR` (with yielding implemented). The queue is 16 elements long and contains elements with sizeof == 4 (pointers).

koobest, 2018-04-02:
Hi, @chanibal ,
could you kindly provide the interrupt handule code(the handler added by gpio_isr_handler_add)？

koobest, 2018-04-08:
Hi, @chanibal, We want to reproduce the problem that you described with gpio's example(frequent triggering of the interrupt),but failed .we don't know exactly what your interrupt handler does. Do you mind providing a simple code that can reproduces this problem? we will find the cause of the problem soon. thanks !!

closed 2018-04-16 by FayeY