espressif/esp-idf #1599: Enable/Disable BT Classic (SPP) issue

LosDeiblos, 2018-02-08:
What is the correct way to disable Bluetooth Classic with SPP, for example for power saving, and then start it up (enable) it again?

I have tried every possible function (esp_bt_controller_disable(), esp_bt_controller_enable(), esp_bluedroid_disable(), esp_bluedroid_enable(), ...) but nothing seams to work. Every time I start up BT Classic with SPP again, after disabling it, the SPP service either doesn't start or I can't write data to the client. Sometimes it also gives me the following Error message: E (436974) BT: bta_jv_disable

blueMoodBHD, 2018-02-11:
@LosDeiblos To disable Bluetooth Classic with SPP, you could run function in turn as follow:
esp_spp_deinit()
esp_bluedroid_disable()
esp_bluedroid_deinit()
esp_bt_controller_disable()
esp_bt_controller_deinit()

And to restart it,you could run function in turn as follow:
esp_bt_controller_init()
esp_bt_controller_enable()
esp_bluedroid_init()
esp_bluedroid_enable()
esp_spp_register_callback()
esp_spp_init()

LosDeiblos, 2018-02-12:
@blueMoodBHD Thanks for the reply. This solved my issue.

closed 2018-02-23 by FayeY