espressif/esp-idf #1601: Espressif's Ethernet board with TLK110 PHY

snowQY, 2018-02-09:
I  found the idf examples ,a ethernet demo ,and   in the readme.md file   has a  sentence  :" The default example configuration is correct for Espressif's Ethernet board with TLK110 PHY." But i found  nothings  in your webpage  about  Espressif's Ethernet board with TLK110 PHY , so can you give me a link or some files about  this board?

sauttefk, 2018-02-09:
This board seems to be no longer available.
Go for a design with LAN8720 (smaller and cheaper)

FayeY, 2018-02-11:
Hi, so sorry that this issue seems to be more like a sales issue, so I'm going to close it. 
You can contact the [ESP sales](https://www.espressif.com/en/company/contact/pre-sale-questions-crm) to ask if there is any ESP's Ethernet board with TLK110 PHY to buy or to apply for.
Thanks for your interest in the ESP32!

closed 2018-02-11 by FayeY