espressif/esp-idf #1604: QIO mode is not enabled for GigaDevice GD25Q32C.

mfp20, 2018-02-09:
I checked the device datasheet and clearly supports QIO. The board's schema shows 4 data lines connected to the device but the bootloader doesn't enable QIO.

In /components/bootloader/subproject/main/flash_qio_mode.c the chip (mfg ID C8, flash ID 4016) is left to the last catch-all entry. Is it the case to add one entry for this device?

negativekelvin, 2018-02-09:
Doesn't catch all lead to 

> Enabling default flash chip QIO

So do you get an error message?

mfp20, 2018-02-09:
Board is the M5Stack's esp32-d0wdq6. Yesterday in wild-search mode I red somewhere about a pin to be pulled high/low during boot to enable QIO ... it may be missing on that board so the default chip's detected mode is not QIO, but the board is capable of QIO. How can I force QIO in order to test the board in QIO mode?

Adding the last row like this maybe?

```
const static qio_info_t chip_data[] = {
/*   Manufacturer,   mfg_id, flash_id, id mask, Read Status,                Write Status,          QIE Bit */
    { "MXIC",        0xC2,   0x2000, 0xFF00,    read_status_8b_rdsr,        write_status_8b_wrsr,  6 },
    { "ISSI",        0x9D,   0x4000, 0xCF00,    read_status_8b_rdsr,        write_status_8b_wrsr,  6 }, /* IDs 0x40xx, 0x70xx */
    { "WinBond",     0xEF,   0x4000, 0xFF00,    read_status_16b_rdsr_rdsr2, write_status_16b_wrsr, 9 },
    { "GD",          0xC8,   0x6000, 0xFF00,    read_status_16b_rdsr_rdsr2, write_status_16b_wrsr, 9 },
    { "GD2",          0xC8,   0x4016, 0xFF00,    read_status_16b_rdsr_rdsr2, write_status_16b_wrsr, 9 },
```
what's the 'id mask' for?

negativekelvin, 2018-02-09:
You should not have to modify the table.  

https://github.com/espressif/esp-idf/blob/4745895e229d8e1ca4e6f4b601397c0ba0b3d2db/components/esptool_py/Kconfig.projbuild#L49-L64

Did you select QIO and how do check if it is enabled or not?

mfp20, 2018-02-09:
Selected QIO, built, flashed, and on reboot the bootloader was saying DIO. I use the esp-idf monitor utility to tap in serial console. I tried both soft and hard (unplug power) reboots and the bootloader always say DIO.

negativekelvin, 2018-02-09:
First stage rom bootloader always uses dio then second stage should switch to qio

mfp20, 2018-02-09:
I can't check by now. I'll try it again as soon as my code is ready to reflash; I can't build it at the moment.
How can I check properly the mode properly?

negativekelvin, 2018-02-09:
Build and flash bootloader with debug log level and look for log

> Enabling QIO mode...

mfp20, 2018-02-10:
My bad... 

``
D (36) qio_mode: Probing for QIO mode enable...
D (41) qio_mode: Raw SPI flash chip id 0x1640c8
D (45) qio_mode: Manufacturer ID 0xc8 chip ID 0x4016
I (50) qio_mode: Enabling default flash chip QIO
D (56) qio_mode: Initial flash chip status 0x2
D (60) qio_mode: QIO mode already enabled in flash
D (65) qio_mode: Enabling QIO mode...
D (69) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (76) boot: magic e9
D (78) boot: segments 04
D (81) boot: spi_mode 02
D (83) boot: spi_speed 00
D (86) boot: spi_size 02
I (88) boot: SPI Speed      : 40MHz
I (93) boot: SPI Mode       : QIO
I (97) boot: SPI Flash Size : 4MB
``
Thanks for helping. How do I mark it 'solved'? Do I have to?

P.s.: I have some spare W25Q128FVSG (Winbond 16MB), do you have any experience with those? From the datasheet looks like I can replace the GD (4MB) one, keep the QIO and have 16MB instead of 4MB.

FayeY, 2018-02-23:
Hi, sorry that we do not have any experience with the W25Q128FVSG (Winbond 16MB). It would be appreciated if you could share it if you have tested it. 
I'm going to close this issue since it is solved. 
Thanks for your interest in the ESP32!

closed 2018-02-23 by FayeY

lgnativs, 2019-05-09:
GD25Q64C QIO support
add item to chip_data[]
`{ "GD",          0xC8,   0x4017 0xFF00,    read_status_8b_rdsr2, write_status_8b_wrsr2,     1 },`

modify the function enable_qio_mode() because must wait #CS high after WREN command

**GD25Q64 datasheet:
7.5. Write Status Register (WRSR) (01H or 31H or 11H)
CS must be driven high after the eighth bit of the data byte has been latched in. If not, the Write
Status Register (WRSR) command is not executed.**

    execute_flash_command(CMD_WREN, 0, 0, 0); 
    esp_rom_spiflash_wait_idle(&g_rom_flashchip);   // wait cs to high
    write_status_fn(status | (1<<status_qio_bit));
    esp_rom_spiflash_wait_idle(&g_rom_flashchip);
    status = read_status_fn();

FayeY, 2019-05-10:
@lgnativs  最后一个值有点问题，改成如下看是否还有问题：
`{ "GD25Q64C", 0xC8, 0x4017 0xFFFF, read_status_8b_rdsr2, write_status_8b_wrsr2, 9 },`

lgnativs, 2019-05-10:
> @lgnativs 最后一个值有点问题，改成如下看是否还有问题：
> `{ "GD25Q64C", 0xC8, 0x4017 0xFFFF, read_status_8b_rdsr2, write_status_8b_wrsr2, 9 },`

rdsr2/wrsr2指令读取一个字节8bit,对应状态寄存器S8-S15,而QE是S9,此时的位移是1
如果使用read_status_16b_rdsr_rdsr2/write_status_16b_wrsr,读取的是一个字16bit,QE是S9,此时的位移是9

lgnativs, 2019-05-10:
在ESP32-WROOM-8MB模块上已经测试成功了

    rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
    configsip: 0, SPIWP:0xee
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:2
    load:0x3fff0018,len:4
    load:0x3fff001c,len:6552
    load:0x40078000,len:10412
    load:0x40080400,len:7116
    entry 0x40080744
    I (28) boot: ESP-IDF v3.3-beta1-268-g5c88c5996 2nd stage bootloader
    I (29) boot: compile time 15:28:09
    I (29) boot: Enabling RNG early entropy source...
    I (35) qio_mode: Enabling QIO for flash chip GD
    I (40) boot: SPI Speed      : 40MHz
    I (44) boot: SPI Mode       : QIO
    I (48) boot: SPI Flash Size : 8MB
    I (52) boot: Partition Table:
    I (56) boot: ## Label            Usage          Type ST Offset   Length
    I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
    I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
    I (78) boot:  2 factory          factory app      00 00 00010000 00180000
    I (85) boot: End of partition table