espressif/esp-idf #1605: An option to place all bss data into external ram would be nice

bernd-edlinger, 2018-02-09:


igrr, 2018-02-12:
Hi @bernd-edlinger,
It is probably not entirely feasible to place all the .bss into external RAM, since .bss is used during application startup when external RAM is not initialized yet. In addition to that, .bss is used for a variety of small static variables such as spinlocks (portMUX_TYPE), flags, and RTOS data structures. Placing these into external RAM would have a very significant performance impact.

It is possible to add a section attribute (similar to RTC_DATA_ATTR or IRAM_ATTR) which would allow placing certain static variable or an array into external RAM. Would that be useful to you?

bernd-edlinger, 2018-02-12:
Hi @igrr,
yes, I think such an attribute would be useful.
Thanks.

bernd-edlinger, 2018-11-19:
Cool.   Thanks a lot!!!