espressif/esp-idf #1608: [Question/Issue]: WPS in APSTA mode

farizadp, 2018-02-09:
Hello,

Has anyone ever managed to get WPS working on mode APSTA?
It seems to me that WPS is functioning on WIFI_MODE_STA only.

Regards,
Dion

FayeY, 2018-02-12:
Hi, yes, it works for sta only, you can find the note in the readme. [WPS Example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/wps)

farizadp, 2018-02-12:
@FayeY : Thanks, Is there any intention in the near-future to get the WPS working on APSTA mode ?

liuzfesp, 2018-02-22:
Hi @farizadp, in idfv3.1, we don't have plan to support it since we already have a lot of work to do. I have put it into our to-do repository, will let you know once we have scheduled it. BTW, why need to run WPS in APSTA mode?

farizadp, 2018-02-23:
Hi @liuzfesp . Thanks for the answer.
Here is a bit of explanation about my ESP32 application:

My ESP32 board hosts a web (AP mode). The web shows its connection status to WLAN router (STA mode). From the web (AP mode), I would like to see the STA mode connection status being updated when WPS is initiated by pressing the physical button on ESP32 board. I therefore need APSTA mode and was thinking that WPS feature on STA mode also works on APSTA mode, which apparently not.

XinDeng11, 2018-03-23:
Hi @farizadp . We have supported the enrollee in APSTA mode, you can check the commit 6c3bdd45 for detail, or you can pull the latest code after it merged.

farizadp, 2018-03-23:
Thanks, @XinDeng11. I couldn't find 6c3bdd45 commit detail. Where can I find it? Or should I really wait until it's merged to master?

closed 2018-04-04 by FayeY