espressif/esp-idf #1610: Problems writing to flash with some API commands but not others.

lonerzzz, 2018-02-11:
We recently received a shipment of manufactured boards that use the ESP-WROOM-32 module. The packaging of this module is different from the previous ESP-WROOM-32 that we had been using during development with markings for CE and FCC certifications now present. We are able write code to this module and to write to SPIFFs using the updater tool so flash writes are possible.

However, we are unable to either read or write successfully using the EEPROM library. With both read and write, no errors are generated and the esp_partition_write call in the commit() function is returning a 0/ESP_OK response. No errors occur, but subsequent attempts to read the flash just written always read as high bits giving 0xFF bytes and 0xFFFF shorts for example.

Is there a need for a new type of module definition with different partitions for the new ESP-WROOM-32 module? Currently the "ESP32 Dev Module" is the type we have been using.

Are there changes in IDF version that may have affected this?

Is any special control of flash read/write rates required? The chip prints mode:DIO on startup.

As a reference, the code works without issue on the ESP-WROOM-32 modules we used for development.

Unfortunately we are totally stuck without the ability to write to flash from code.

UPDATE: We have now confirmed that the esp_partition_write/esp_partition_read methods do not consistently write to the flash. Rarely, the flash will be updated with a few bytes, but in most cases it is not updated by these methods. In contrast, the nvs_set_X/nvs_get_X methods seem to work fine.

igrr, 2018-02-11:
Could you please check and mention:

1) ESP-IDF commit ID which you are using

2) Check the flash chip vendor and ID using esptool.py:
```
$IDF_PATHcomponents/esptool_py/esptool/esptool.py --port YOUR_PORT_NAME flash_id
```

3) Use `make menuconfig` to enable "Verify SPI flash writes" option under "Component config", "SPI flash driver". The rebuild the application and see if you get any flash write verification errors on the console. Log level (set under "Component config", "Log output" should be set at least to "Warning"

lonerzzz, 2018-02-11:
Based on the information here, https://github.com/espressif/arduino-esp32, the IDF commit id is 1c3dd23. We picked up the latest code. We did not correlate this back to an official revision. It is was @me_no_dev put in if I understand correctly. The SDK reported the ESP API is v3.1-dev-239-g1c3dd23f-dirty.

The output of the esptool command is:
esptool.py v2.1
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4016
Detected flash size: 4MB
Hard resetting...

Will try the verify SPI flash writes option.

Thanks,

Jason

igrr, 2018-02-11:
Could you also please share the test case you have for partition read/write methods? If you can't share it publicly, could you email it to ivan at espressif com?

lonerzzz, 2018-02-11:
Other code was trying to use the EEPROM and SPIFFS functionality, but since that was not working, the following test was made which reproduces the error. Reads seem to work in this model as the same data is always returned, but writes don't seem to be functional consistently.
```
 uint8_t data[ 512 ];
  for (uint16_t i=0; i<512; i++)
  {
    data[ i ] = i;
  }
  const esp_partition_t* partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,ESP_PARTITION_SUBTYPE_ANY, "eeprom");
  esp_partition_write( partition, 0, data, 512 );
  for (uint16_t i=0; i<512; i++)
  {
    data[ i ] = 0x55;
  }
  esp_partition_read( partition, 0, data, 512 );
  for (uint16_t i=0; i<512; i++)
  {
    Serial.printf( " %02x", data[ i ] );
    if ((i & 0x000F) == 0x000F)
    {
      Serial.println("");
    }
  }
```

lonerzzz, 2018-02-12:
After taking a sample of 10 units, we determined that we have write errors on 4 of them so we seem to have a lot of poor quality modules. On the ones that seem to work, the EEPROM code works, but we still have issues with the SPIFFS as we can see files but they cannot be read. Continuing to dig... Thanks for the help.

lonerzzz, 2018-02-12:
The issue with SPIFFS is separate and due to a bad version of (mkspiffs) in the load as per (https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/6) Unfortunately, that is the one now active in the Arduino project.

copercini, 2018-02-12:
@lonerzzz  your SPIFFS problem looks like this one: https://github.com/espressif/arduino-esp32/issues/1022

igrr, 2018-02-12:
@lonerzzz could you please try mkspiffs from https://github.com/igrr/mkspiffs/releases/tag/0.2.2? Download the release which has "arduino-esp32" in its name.

lonerzzz, 2018-02-12:
@igrr Your latest version works with the latest code in the Arduino project. Thanks for the reference. I have updated the other bugs until @me-no-dev spins a new version including your update.

igrr, 2018-02-13:
@lonerzzz Thanks for testing. I'm closing this issue since it appears to be solved. Please reopen if there is anything else which needs to be fixed.

closed 2018-02-13 by igrr

baggior, 2018-02-17:
not solved for me
mkspiffs arduino  v 2.2.0
reports 
'SPIFFS mount failed with error: -10025'

igrr, 2018-02-17:
@baggior You can safely ignore this error.
It was reported in https://github.com/igrr/mkspiffs/issues/39 and fixed in latest git version of mkspiffs.

TD-er, 2018-07-08:
In what PlatformIO version is this fixed?
Using the latest ESP32 version with PlatformIO, I have a lot of issues reading back what has been written by that version.
Data written by a previous version is read just fine.