espressif/esp-idf #1611: Cannot/(Howto?) Add %.ext Pattern Rule To component.mk

paulreimer, 2018-02-11:
This pattern rule works within a `component.mk` file for its own files, but does not work project-wide, if put in a `Makefile.projbuild`.

```
%_generated.h: %.fbs
       flatc --cpp -o $(@D) $^
```

To be clear, I believe this works but I need to put the rule in every component where it is used. The effect I want is to write it once and then any "*.fbs" file will be used to produce a "*_generated.h" file.

e.g. In a different `component.mk` I want to use it like this:
```
foobar_test_task.o: $(COMPONENT_PATH)/foobar_generated.h
```

It produces the following error:
```
make[1]: *** No rule to make target `/PATH/TO/APP/main/firebase_db_generated.h', needed by `foobar_test_task.o'.  Stop.
```

paulreimer, 2018-02-11:
I've tried a few combinations of prepending `$(COMPONENT_PATH)` or `$$(COMPONENT_PATH)` to the targets and/or dependencies, in either the pattern rule or in the other component, and tried the pattern rule in `component.mk` and/or `Makefile.proj`.

igrr, 2018-02-11:
Perhaps you could try adding this rule to Makefile.componentbuild? http://esp-idf.readthedocs.io/en/latest/api-guides/build-system.html#makefile-componentbuild

paulreimer, 2018-02-11:
Thanks! Of course, that works perfectly.

Is `COMPONENT_CONFIG_ONLY` appropriate in this case, e.g. there is a component which provides a rule in `Makefile.componentbuild`, and also contains some header-only directories (but nothing to be compiled)?

As well (related but not strictly to this issue), when adding a dependency via a `component.mk` object override, how to combine both additional prerequisites and also to override e.g. CFLAGS?

```
foobar_test_task.o: \
  $(COMPONENT_PATH)/foobar_generated.h \
  CXXFLAGS += -D_GLIBCXX_USE_C99=1

# make[1]: *** No rule to make target `CXXFLAGS', needed by `rxcpp_test_task.o'.  Stop.
```

```
foobar_test_task.o: \
  CXXFLAGS += -D_GLIBCXX_USE_C99=1 \
  $(COMPONENT_PATH)/foobar_generated.h

# xtensa-esp32-elf-c++: fatal error: cannot specify -o with -c, -S or -E with multiple files
```

paulreimer, 2018-02-11:
Well, to answer that second question it seems just override it twice and they get merged:
```
foobar_test_task.o: $(COMPONENT_PATH)/foobar_generated.h
foobar_test_task.o: CXXFLAGS += -D_GLIBCXX_USE_C99=1
```

So, only remaining question is whether `COMPONENT_CONFIG_ONLY` is suitable for a header-only library w/`Makefile.componentbuild` including a pattern rule as discussed here.

paulreimer, 2018-02-11:
At a glance in `make/component_wrapper.mk` it appears that `COMPONENT_CONFIG_ONLY` won't set `COMPONENT_INCLUDES`. I'm not sure if there is a preferred way to omit building a library, but still add to project-wide headers (should I set `COMPONENT_CONFIG_ONLY` and modify `CXXFLAGS` directly (`-I $(COMPONENT_PATH)/`) in a `Makefile.projbuild`. Or is it better to create a dummy `COMPONENT_OWNBUILDTARGET`...?)

igrr, 2018-02-12:
Usually we omit building the library by adding

```
COMPONENT_SRCDIRS :=
COMPONENT_ADD_LDFLAGS :=
```

(which means: don't compile anything, don't link anything).

FayeY, 2018-02-23:
Hi @paulreimer , did the above suggestion help you resolve the issue?

closed 2018-03-20 by igrr

paulreimer, 2018-03-27:
Hi @FayeY,

Sorry for the delay -- and I see the issue was closed -- I had a bit of time to resolve the issue today and I was able to use the suggestions by @igrr as well as the (undocumented?) variable  `COMPILING_COMPONENT_PATH` which was the missing part of the solution.

Thanks all for the help!