espressif/esp-idf #1612: Ble characteristic length incorrect after long writes

jfdevmoto, 2018-02-11:
see https://github.com/espressif/esp-idf/issues/771
I think there may be a regression on this issue fix.
I have a problem after long writes. It looks like the lenght value is doubled or at least is badly updated.
I have a 159 bytes long characteristic. Then my gatt client (android APP) writes 159 bytes. 
Suddenly during the last GATT_WRITE_EVT callback processing, the lenght value becomes 318. Then I receive the ESP_GATTS_EXEC_WRITE_EVT.
control is with autoresponse (ESP_GATT_AUTO_RSP).
I think there is a regression in the code as latest code varies significantly from what is written in above mentionned issue.
line 425 may have to be changed to
queue_data->p_attr->p_value->attr_val.attr_len = queue_data->len;
instead of
queue_data->p_attr->p_value->attr_val.attr_len += queue_data->len;

Can you confirm ? I don't have much time to look deeper and validate. Would be great for you to confirm before I come back from holliday.

If this is not the issue, please give me some other help.
I am working with esp-idf 3.1 dev168
Thanks

Weijian-Espressif, 2018-02-12:
@jfdevmoto I test gatt_client demo and gatt_server demo with prepare write, it is ok.  And I test nRF connect APP prepare write to gatt_server demo in ESP32, it is ok too.  What is the commit ID you used?

jfdevmoto, 2018-02-12:
Thanks for your reply.
As I may have noticed (or not ?) I use automatic control for this
characteristic. So I think this is not the same as in gatts demo. Of course
is my app is controling the data value read write, there would be no issue.
I am out of office for two weeks so I can't investigate further until I
come back.

Weijian-Espressif, 2018-02-13:
@jfdevmoto 👌 Happy holidays

Weijian-Espressif, 2018-02-13:
@jfdevmoto  our gatt_server_service_table demo is automatic control . it will send auto response when you write or prepare write. you can test this demo , in line 41 , GATTS_DEMO_CHAR_VAL_LEN_MAX is the the max length of char value, the length of you write or prepare wrie must less than GATTS_DEMO_CHAR_VAL_LEN_MAX, so i suggest you increase the value of GATTS_DEMO_CHAR_VAL_LEN_MAX.

Weijian-Espressif, 2018-02-24:
@jfdevmoto Sorry, this is a bug. I will fix the bug in the latest IDF. The fix code below
**line 418 in gatt_process_exec_write_req of gatt_sr.c**
BOOLEAN is_first = TRUE;
    //dequeue prepare write data
    while(fixed_queue_try_peek_first(prepare_record->queue)) {
        queue_data = fixed_queue_dequeue(prepare_record->queue);
        if (is_prepare_write_valid){
            if((queue_data->p_attr->p_value != NULL) && (queue_data->p_attr->p_value->attr_val.attr_val != NULL)){
                if(is_first) {
                    //clear attr_val.attr_len before handle prepare write data
                    queue_data->p_attr->p_value->attr_val.attr_len = 0;
                    is_first = FALSE;
                }
                memcpy(queue_data->p_attr->p_value->attr_val.attr_val+queue_data->offset, queue_data->value, queue_data->len);
                //don't forget to increase the attribute value length in the gatts database.
                queue_data->p_attr->p_value->attr_val.attr_len += queue_data->len;
            }
        }
        osi_free(queue_data);
    }

jfdevmoto, 2018-02-25:
Thanks a lot for your concern and corrective action.

closed 2018-02-25 by jfdevmoto