espressif/esp-idf #1613: Crash when trying to free BT classic memory with blufi example

abhimanyuv1, 2018-02-12:
When trying to free up BT classic memory in blufi example i get the following crash. Is it expected that BT classic should be present when using blufi profile? My understanding is that it is BTLE profile. Please let me understand if it is real issue or i am incorrectly using it.

Code:
void ble_task(void *pvParameters) {
    esp_err_t ret;

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGI(TAG, "Initialize bt controller failed\n");
        return;
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGI(TAG, "Enable bt controller failed\n");
        return;
    }

    // Release classic BT memory
    esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);

    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGI(TAG, "Init bluedroid failed\n");
        return;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGI(TAG, "Init bluedroid failed\n");
        return;
    }

    ESP_LOGI(TAG, "BD ADDR: "ESP_BD_ADDR_STR"\n", ESP_BD_ADDR_HEX(esp_bt_dev_get_address()));
    ESP_LOGI(TAG, "BLUFI VERSION %04x\n", esp_blufi_get_version());

    blufi_security_init();
    esp_ble_gap_register_callback(ble_gap_event_handler);

    esp_blufi_register_callbacks(&blufi_event_callback);
    esp_blufi_profile_init();

    vTaskDelete(NULL);
}

Issue:

Stack trace:
esp-idf/components/freertos/./queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40089389 on core 0
0x40089389: xQueueGenericReceive at /esp-idf/components/freertos/./queue.c:2520

Backtrace: 0x4008c9e0:0x3ffb5370 0x4008caab:0x3ffb5390 0x40089389:0x3ffb53b0 0x40151171:0x3ffb53f0 0x40168b7f:0x3ffb5410 0x40176ece:0x3ffb5430 0x40176fae:0x3ffb5450 0x40173770:0x3ffb5480 0x4016d22d:0x3ffb54b0 0x4016ab09:0x3ffb54d0 0x4018aab5:0x3ffb54f0 0x4017f4d2:0x3ffb5510 0x4017fab1:0x3ffb5530 0x40168ae6:0x3ffb5550
0x4008c9e0: invoke_abort at /esp-idf/components/esp32/./panic.c:577

0x4008caab: abort at /esp-idf/components/esp32/./panic.c:577

0x40089389: xQueueGenericReceive at /esp-idf/components/freertos/./queue.c:2520

0x40151171: osi_mutex_lock at /esp-idf/components/bt/bluedroid/osi/mutex.c:49

0x40168b7f: btu_start_timer at /esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:559

0x40176ece: l2cu_no_dynamic_ccbs at /esp-idf/components/bt/bluedroid/stack/l2cap/l2c_utils.c:1821 (discriminator 2)

0x40176fae: l2cu_release_ccb at /esp-idf/components/bt/bluedroid/stack/l2cap/l2c_utils.c:1821 (discriminator 2)

0x40173770: L2CA_RemoveFixedChnl at /esp-idf/components/bt/bluedroid/stack/l2cap/l2c_api.c:1902

0x4016d22d: gatt_disconnect at /esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:1178

0x4016ab09: GATT_Deregister at /esp-idf/components/bt/bluedroid/stack/gatt/gatt_api.c:1268

0x4018aab5: bta_gatts_api_disable at /esp-idf/components/bt/bluedroid/bta/gatt/bta_gatts_act.c:531

0x4017f4d2: bta_gatts_hdl_event at /esp-idf/components/bt/bluedroid/bta/gatt/bta_gatts_main.c:68

0x4017fab1: bta_sys_event at /esp-idf/components/bt/bluedroid/bta/sys/bta_sys_main.c:631

0x40168ae6: btu_task_thread_handler at /esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:559

Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG crosstool-ng-1.22.0-75-gbaf03c2) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /esp32app/build/app.elf...done.
Remote debugging using /dev/ttyUSB0
0x4008c9e0 in invoke_abort () at /esp-idf/components/esp32/./panic.c:141
141	        *((int *) 0) = 0;
(gdb) bt
#0  0x4008c9e0 in invoke_abort () at /esp-idf/components/esp32/./panic.c:141
#1  0x4008caae in abort () at /esp-idf/components/esp32/./panic.c:150
#2  0x4008938c in xQueueGenericReceive (xQueue=0x3ffbd970, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /esp-idf/components/freertos/./queue.c:1442
#3  0x40151174 in osi_mutex_lock (mutex=0x3ffd76ec <btu_general_alarm_lock>, timeout=4294967295) at /esp-idf/components/bt/bluedroid/osi/mutex.c:49
#4  0x40168b82 in btu_start_timer (p_tle=0x3ffd58d4 <l2cb+20>, type=2, timeout_sec=30) at /esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:452
#5  0x40176ed1 in l2cu_no_dynamic_ccbs (p_lcb=0x3ffd58cc <l2cb+12>) at /esp-idf/components/bt/bluedroid/stack/l2cap/l2c_utils.c:2737
#6  0x40176fb1 in l2cu_release_ccb (p_ccb=<optimized out>) at /esp-idf/components/bt/bluedroid/stack/l2cap/l2c_utils.c:1683
#7  0x40173773 in L2CA_RemoveFixedChnl (fixed_cid=<optimized out>, rem_bda=<optimized out>) at /esp-idf/components/bt/bluedroid/stack/l2cap/l2c_api.c:1902
#8  0x4016d230 in gatt_disconnect (p_tcb=0x3ffd45c0 <gatt_cb>) at /esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:243
#9  0x4016ab0c in GATT_Deregister (gatt_if=3 '\003') at /esp-idf/components/bt/bluedroid/stack/gatt/gatt_api.c:1268
#10 0x4018aab8 in bta_gatts_api_disable (p_cb=0x3ffd72ac <bta_gatts_cb>) at /esp-idf/components/bt/bluedroid/bta/gatt/bta_gatts_act.c:152
#11 0x4017f4d5 in bta_gatts_hdl_event (p_msg=0x3ffb668c) at /esp-idf/components/bt/bluedroid/bta/gatt/bta_gatts_main.c:68
#12 0x4017fab4 in bta_sys_event (p_msg=0x3ffb668c) at /esp-idf/components/bt/bluedroid/bta/sys/bta_sys_main.c:491
#13 0x40168ae9 in btu_task_thread_handler (arg=<optimized out>) at /esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:230

chegewara, 2018-02-12:
https://github.com/espressif/esp-idf/blob/5401a75badcfc9c20f7818fdf3c683e3da8b2547/components/bt/include/esp_bt.h#L215-L216

abhimanyuv1, 2018-02-13:
Thank you chegewara. I must have read it incorrectly.

closed 2018-02-13 by abhimanyuv1