espressif/esp-idf #1614: I2S driver for internal 8 bit DAC is broken

pufstudio, 2018-02-12:
I can't get internal DAC to produce sound when using the a2dp_sink example.

This used to work when I first tested it around September 2017.
I have been using external I2S DAC ever since so I'm not sure when the issue with internal DAC started.

I sometimes get crackling noises but music can't be heard.

I noticed that calculated real rate is way off the target:
> I (192) I2S: PLL_D2: Req RATE: 44100, real rate: 2777.000, BITS: 16, CLKM: 30, BCK: 60, MCLK: 30.234, SCLK: 88864.000000, diva: 64, divb: 14

If I set .use_apll = 1 the output changes to:
> I (204) I2S: APLL: Req RATE: 44100, real rate: 44099.020, BITS: 16, CLKM: 1, BCK: 8, MCLK: 30.234, SCLK: 1411168.625000, diva: 1, divb: 0

But still it's not working.

koobest, 2018-03-06:
Hi, @pufstudio, we can not use apll clk in i2s adc mode.

pufstudio, 2018-03-06:
Thanks for your reply koobest, but it's also not working without apll.

I'm using a2dp_sink example without any modification (except setting to internal dac in menuconfig) with the esp-idf from master branch.

Looks to me like sample rate is calculated wrong but maybe that's how it's supposed to be...
When it was working back in the day I haven't actually been paying close attention to the "real rate".

koobest, 2018-03-12:
Hi, pufstudio, Where is a2dp_sink example, I didn't find it example and recently we have found that opening wifi affects I2s ADC.  Have you used wifi in your code?

negativekelvin, 2018-03-12:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_sink

pufstudio, 2018-03-12:
@koobest: No. I'm not using WiFi with the example. Only classic Bluetooth.

koobest, 2018-03-13:
@pufstudio 
I have a test with this example, but no sound can be heard. so, I change  `communication_format` to
`communication_format = I2S_COMM_FORMAT_I2S_MSB`. music can be heard but not so clear.
because the DAC is 8 bit out put.

pufstudio, 2018-03-13:
@koobest 
Thank you. That was certainly part of the issue.
I found out that .mode also needs to be set to I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN instead of just I2S_MODE_DAC_BUILT_IN as it's set in the example.

FayeY, 2018-03-20:
Yes, my mistake, sorry for that.

FayeY, 2018-05-22:
The example has been updated, please have a try. Thanks.

closed 2018-05-22 by FayeY

macaba, 2018-10-24:
Note for anyone else who came across this issue; to resolve it for my hardware, I had to modify `bt_app_a2d_data_cb` to:
```
uint16_t buffer[2048];
void bt_app_a2d_data_cb(const uint8_t *data, uint32_t len)
{
    size_t bytes_written;
    for(int i = 0; i < 2048; i++){
        buffer[i] = ((int16_t*)data)[i] + 0x8000;
    }
    i2s_write(0, (const uint8_t*)buffer, len, &bytes_written, portMAX_DELAY);
    if (++s_pkt_cnt % 100 == 0) {
        ESP_LOGI(BT_AV_TAG, "Audio packet count %u", s_pkt_cnt);
    }
}
```

The important part was to add the offset of ` 0x8000`. Hope this helps.