espressif/esp-idf #1615: Removing or replacing IDF components

ammaree, 2018-02-12:
couple of questions, but first some background

Since our devices are built to run completely headless (no UART console) and since the UART port is used for completely different purposes, we would like to redirect all ESP_LOG?() output to a circular buffer accessed via a secured (username/password) telnet session. We also have the functionality in our syslog() function to log all messages below a system defined severity level to a full function syslog server.

Whilst we have got this (sort of) working by just writing a small replacement esp_log_write() function and commenting out the original function this is not ideal for a number of reasons:
a) The mechanism only works with WARNING and higher (and not DEBUG) and we are having difficulty identifying why.
b) The message, when it reaches our esp_log_write() function, is already semi-formatted with tags that we end up duplicating to create a proper RFC5424 formatted message.
c) We regularly have to revert and redo the log.c file changes, mainly a nuisance.

Q1: How can we disable the inclusion of a module such as the log component completely and persistently in order to replace it with our syslog implementation?

Q2: Any hints as to why the mechanism does not work with DEBUG level?
 
The answer to Q1 will also help us to remove conflict caused in Eclipse between the exact duplicated definitions in the AWS client in the IDF and the native Paho Embedded C client we are using to communicate with our host.

Andre

igrr, 2018-02-12:
> How can we disable the inclusion of a module such as the log component completely and persistently in order to replace it with our syslog implementation?

You can override any ESP-IDF component by adding a component with the same name in your project's components directory. This is described here: http://esp-idf.readthedocs.io/en/latest/api-guides/build-system.html#multiple-components-with-the-same-name

You will still have to provide a header file compatible with esp_log.h, since other components depend on it. However you are free to customize the implementation of ESP_LOGx macros, i.e. you may not even need esp_log_write in this case.

> Any hints as to why the mechanism does not work with ERROR level?

I don't have any clue, sorry. I know of a few applications which use esp_log_set_vprintf to redirect formatted messages to a different location (such as UDP or ESP-NOW), and all messages are redirected correctly. This also means that these messages are being passed to esp_log_write.

If you can reduce your application to a small part which can be shared and which is sufficient to reproduce the issue, we'll certainly have a look.

ammaree, 2018-02-13:
Hi @igrr thanks for the feedback.

I did as suggested but running into some problems, I think related to my makefile.
I duplicated the whole $(IDF_PATH)/components/log to my directory $(PATH_COMPONENTS)/log and made the necessary changes, but the modified module is not being used.

Believing the problems might be related to the used of the deprecated SRCDIRS variable I changed that to EXTRA_COMPONENT_DIRS and then ran into an error relating to "freertos/FReeRTOS.h: No such file or directory"

I then tried to fix that by adding "-I $(IDF_PATH)/components/freertos/include" to the CFLAGS variable but still no luck, any suggestion. My directory structure is as follows:

C:/Dropbox/devs/ws/z-sdk/esp-idf
                                 /z-appl/ewm-irmacos {application .c & .h}
                                 /z-components/log {my syslog .c & .h}
                                                         /many other components...

My makefile is as follows:
`# This is a project Makefile. It is assumed the directory this Makefile resides in is a project subdirectory.
BUILD_MODEL						:=	3
ifeq ($(BUILD_MODEL), 1)
PLATFORM_MODEL					:=	configESP32_EBOX=1
else ifeq ($(BUILD_MODEL), 2)
PLATFORM_MODEL					:=	configESP32_EWMv0=1
else ifeq ($(BUILD_MODEL), 3)
PLATFORM_MODEL					:=	configESP32_AC_01=1
endif

BIN_FILE_NAME					:=	6
ifeq ($(BIN_FILE_NAME), 1)		#	Site token
PROJECT_NAME					:=	06e29441-a2f7-4b2c-9f89-ebcdf0358fae
else ifeq ($(BIN_FILE_NAME), 2)	#	Specification token
PROJECT_NAME					:=	09d2b3e3-b034-4753-b1fe-13976a47bac4
else ifeq ($(BIN_FILE_NAME), 3)	#	192.168.1.77
PROJECT_NAME					:=	240ac4035de8
else ifeq ($(BIN_FILE_NAME), 4)	#	192.168.1.78
PROJECT_NAME					:=	240ac402d278
else ifeq ($(BIN_FILE_NAME), 5)	#	192.168.1.79
PROJECT_NAME					:=	240ac4035c08
else ifeq ($(BIN_FILE_NAME), 6)	#	192.168.1.80
PROJECT_NAME					:=	30aea432bc00
else ifeq ($(BIN_FILE_NAME), 7)	#	192.168.1.81
PROJECT_NAME					:=	30aea432bbe4
endif

PLATFORM_HAL				:=	esp32
PATH_APPL					:=	/c/Dropbox/devs/ws/z-appl
PATH_COMPONENTS				:=	/c/Dropbox/devs/ws/z-components
PATH_SUPPORT				:=	/c/Dropbox/devs/ws/z-support

# Get the Wifi Firmware version from the git commit ID ( https://github.com/espressif/esp-idf/issues/1472 )
WIFI_VER := $(shell git -C ${IDF_PATH}/components/esp32/lib rev-parse --short=8 HEAD)

# Create value & echo to "timestamp.h" file with compile time for accurate FOTA tests
TIMESTAMP =	$(shell date +"%s")
$(shell echo \#define APP_TIMESTAMP $(TIMESTAMP) > "$(PATH_APPL)/iMacs/timestamp.h")

#SRCDIRS
EXTRA_COMPONENT_DIRS		+=	$(PATH_APPL)/iMacs							\
								$(PATH_COMPONENTS)/actuators				\
								$(PATH_COMPONENTS)/common					\
								$(PATH_COMPONENTS)/endpoints				\
								$(PATH_COMPONENTS)/rtos-support				\
								$(PATH_COMPONENTS)/hal_$(PLATFORM_HAL)		\
								$(PATH_COMPONENTS)/http						\
								$(PATH_COMPONENTS)/identity					\
								$(PATH_COMPONENTS)/log						\
								$(PATH_COMPONENTS)/mqtt						\
								$(PATH_COMPONENTS)/rules					\
								$(PATH_COMPONENTS)/sensors					\
								$(PATH_COMPONENTS)/sitewhere				\
								$(PATH_SUPPORT)/http-parser					\
								$(PATH_SUPPORT)/jsmn						\
								$(PATH_SUPPORT)/nanopb						\
								$(PATH_SUPPORT)/libyuarel

CFLAGS						+=	-I $(PATH_APPL)/iMacs						\
								-I $(PATH_COMPONENTS)/actuators				\
								-I $(PATH_COMPONENTS)/common				\
								-I $(PATH_COMPONENTS)/endpoints				\
								-I $(PATH_COMPONENTS)/rtos-support			\
								-I $(PATH_COMPONENTS)/hal_$(PLATFORM_HAL)	\
								-I $(PATH_COMPONENTS)/http					\
								-I $(PATH_COMPONENTS)/identity				\
								-I $(PATH_COMPONENTS)/log					\
								-I $(PATH_COMPONENTS)/mqtt					\
								-I $(PATH_COMPONENTS)/rules					\
								-I $(PATH_COMPONENTS)/sensors				\
								-I $(PATH_COMPONENTS)/sitewhere				\
								-I $(PATH_SUPPORT)/http-parser				\
								-I $(PATH_SUPPORT)/jsmn						\
								-I $(PATH_SUPPORT)/nanopb					\
								-I $(PATH_SUPPORT)/libyuarel				\
								-I $(IDF_PATH)/components/freertos/include

# Build definitions to be added
CPPFLAGS					+=	-D PB_NO_PACKED_STRUCTS=1					\
								-D PB_BUFFER_ONLY=1							\
								-D $(PLATFORM_MODEL)						\
								-D DEV_WIFI_VER="$(WIFI_VER)" 
								
# Do NOT use LDFLAGS, it overwrites the default flags. With esp-idf-v2.1 we need the $(PROJECT_PATH)/main entry
EXTRA_LDFLAGS				+=	-L $(BUILD_DIR_BASE)/iMacs					\
								-L $(BUILD_DIR_BASE)/actuators				\
								-L $(BUILD_DIR_BASE)/common					\
								-L $(BUILD_DIR_BASE)/endpoints				\
								-L $(BUILD_DIR_BASE)/rtos-support			\
								-L $(BUILD_DIR_BASE)/hal_$(PLATFORM_HAL)	\
								-L $(BUILD_DIR_BASE)/http					\
								-L $(BUILD_DIR_BASE)/identity				\
								-L $(BUILD_DIR_BASE)/log					\
								-L $(BUILD_DIR_BASE)/mqtt					\
								-L $(BUILD_DIR_BASE)/rules					\
								-L $(BUILD_DIR_BASE)/sensors				\
								-L $(BUILD_DIR_BASE)/sitewhere				\
								-L $(BUILD_DIR_BASE)/http-parser			\
								-L $(BUILD_DIR_BASE)/jsmn					\
								-L $(BUILD_DIR_BASE)/nanopb					\
								-L $(BUILD_DIR_BASE)/libyuarel

# MUST BE AT THE END
include $(IDF_PATH)/make/project.mk
`

ammaree, 2018-02-13:
Apologies for the cut/paste but all padding space and trailing "\\" is being removed

igrr, 2018-02-13:
Could you please run `make list-components` in your project and paste the result here?

The makefile indeed looks interesting. Here are a few suggestions (which may not immediately help solve the issue):

1. If you have a directory with a bunch of components, you can set `EXTRA_COMPONENT_DIRS` to that directory. No need to list every single component. For example,
    ```
    EXTRA_COMPONENT_DIRS := $(PATH_COMPONENTS)
    ```
    (Also you don't need to append to EXTRA_COMPONENT_DIRS if you only intend to set it once. `:=` is sufficient.)

2. If component makefiles are written correctly, you most definitely don't need to list all components' directories in `CFLAGS` and `EXTRA_LDFLAGS` of the top level project. Components should use `COMPONENT_ADD_INCLUDEDIRS` to set the include directories which are to be added to include paths when building other components.

3. When adding CFLAGS/CPPFLAGS/CXXFLAGS from the top level project makefile, set `EXTRA_CFLAGS`/`EXTRA_CPPFLAGS`/`EXTRA_LDFLAGS`. This will ensure that you still have all the original flags (determined by the build system) set correctly.

By the way, you use triple backticks to quote blocks of code here, and single backticks when marking inline code. Please refer to https://guides.github.com/features/mastering-markdown/, and check "Preview" tab above to see if the result looks as you expect.

ammaree, 2018-02-13:
Thanks for the advice on only listing the top level directory containing the components, have tried it but caused problems due to fact that the same component structure contain modules (hal and otherwise) related to different platforms and different/conflicting modules, hence selecting only those that are applicable.
 
When I originally pasted the makefile I used the single backticks, but the '#' used in makefiles for comments resulting in big bold headings so I reverted to simple cut & paste.

Result of 'make list-components' pasted below.

Help appreciated.

Andre

'*******************************************************************************
COMPONENT_DIRS (components searched for here)
/c/Dropbox/devs/ws/z-appl/ewm-irmacos/components
/c/Dropbox/devs/ws/z-appl/iMacs
/c/Dropbox/devs/ws/z-components/actuators
/c/Dropbox/devs/ws/z-components/common
/c/Dropbox/devs/ws/z-components/endpoints
/c/Dropbox/devs/ws/z-components/rtos-support
/c/Dropbox/devs/ws/z-components/hal_esp32
/c/Dropbox/devs/ws/z-components/http
/c/Dropbox/devs/ws/z-components/identity
/c/Dropbox/devs/ws/z-components/mqtt
/c/Dropbox/devs/ws/z-components/rules
/c/Dropbox/devs/ws/z-components/sensors
/c/Dropbox/devs/ws/z-components/sitewhere
/c/Dropbox/devs/ws/z-support/http-parser
/c/Dropbox/devs/ws/z-support/jsmn
/c/Dropbox/devs/ws/z-support/nanopb
/c/Dropbox/devs/ws/z-support/libyuarel
/c/Dropbox/devs/ws/z-sdk/esp-idf/components
/c/Dropbox/devs/ws/z-appl/ewm-irmacos/main
*******************************************************************************
COMPONENTS (list of component names)
actuators app_trace app_update aws_iot bootloader bootloader_support bt coap common console cxx driver endpoints esp32 esp_adc_cal esptool_py ethernet expat fatfs freertos hal_esp32 heap http http-parser iMacs identity idf_test jsmn json libsodium libyuarel log lwip mbedtls mdns micro-ecc mqtt nanopb newlib nghttp nvs_flash openssl partition_table pthread rtos-support rules sdmmc sensors sitewhere soc spi_flash spiffs tcpip_adapter ulp vfs wear_levelling wpa_supplicant xtensa-debug-module
*******************************************************************************
COMPONENT_PATHS (paths to all components):
/c/Dropbox/devs/ws/z-components/actuators
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/app_trace
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/app_update
/c/Dropbox/devs/ws/z-components/aws_iot
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/bootloader
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/bootloader_support
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/bt
/c/Dropbox/devs/ws/z-components/coap
/c/Dropbox/devs/ws/z-components/common
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/console
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/cxx
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/driver
/c/Dropbox/devs/ws/z-components/endpoints
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/esp32
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/esp_adc_cal
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/esptool_py
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/ethernet
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/expat
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/fatfs
/c/Dropbox/devs/ws/z-support/freertos
/c/Dropbox/devs/ws/z-components/hal_esp32
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/heap
/c/Dropbox/devs/ws/z-components/http
/c/Dropbox/devs/ws/z-support/http-parser
/c/Dropbox/devs/ws/z-appl/iMacs
/c/Dropbox/devs/ws/z-components/identity
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/idf_test
/c/Dropbox/devs/ws/z-support/jsmn
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/json
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/libsodium
/c/Dropbox/devs/ws/z-support/libyuarel
/c/Dropbox/devs/ws/z-components/log
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/lwip
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/mbedtls
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/mdns
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/micro-ecc
/c/Dropbox/devs/ws/z-components/mqtt
/c/Dropbox/devs/ws/z-support/nanopb
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/newlib
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/nghttp
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/nvs_flash
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/openssl
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/partition_table
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/pthread
/c/Dropbox/devs/ws/z-components/rtos-support
/c/Dropbox/devs/ws/z-components/rules
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/sdmmc
/c/Dropbox/devs/ws/z-components/sensors
/c/Dropbox/devs/ws/z-components/sitewhere
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/soc
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/spi_flash
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/spiffs
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/tcpip_adapter
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/ulp
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/vfs
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/wear_levelling
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/wpa_supplicant
/c/Dropbox/devs/ws/z-sdk/esp-idf/components/xtensa-debug-module
'

ammaree, 2018-02-13:
And here is the make output/errors.

'C:/Dropbox/devs/ws/z-sdk/esp-idf/components/app_update/esp_ota_ops.c:22:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
In file included from C:/Dropbox/devs/ws/z-sdk/esp-idf/components/app_trace/include/esp_app_trace.h:19:0,
                 from C:/Dropbox/devs/ws/z-sdk/esp-idf/components/app_trace/host_file_io.c:26:
C:/Dropbox/devs/ws/z-sdk/esp-idf/components/app_trace/include/esp_app_trace_util.h:17:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
make[1]: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/component_wrapper.mk:274: esp_ota_ops.o] Error 1
make[1]: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/component_wrapper.mk:274: host_file_io.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from C:/Dropbox/devs/ws/z-components/hal_esp32/hal_usart.h:7:0,
                 from C:/Dropbox/devs/ws/z-components/common/x_stdio.h:27,
                 from C:/Dropbox/devs/ws/z-components/common/x_printf.h:91,
                 from C:/Dropbox/devs/ws/z-components/common/x_debug.h:31,
                 from C:/Dropbox/devs/ws/z-components/common/x_time.c:25:
C:/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/include/driver/uart.h:28:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
make: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/project.mk:450: component-app_update-build] Error 2
make: *** Waiting for unfinished jobs....
C:/Dropbox/devs/ws/z-sdk/esp-idf/components/app_trace/app_trace.c:165:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
C:/Dropbox/devs/ws/z-sdk/esp-idf/components/bt/bt.c:21:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
In file included from C:/Dropbox/devs/ws/z-components/hal_esp32/hal_i2c.h:10:0,
                 from C:/Dropbox/devs/ws/z-components/actuators/actuators.h:12,
                 from C:/Dropbox/devs/ws/z-components/actuators/actuators.c:7:
C:/Dropbox/devs/ws/z-sdk/esp-idf/components/driver/include/driver/i2c.h:25:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
make[1]: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/component_wrapper.mk:274: x_time.o] Error 1
In file included from C:/Dropbox/devs/ws/z-components/hal_esp32/hal_network.h:7:0,
                 from C:/Dropbox/devs/ws/z-components/common/x_network.h:28,
                 from C:/Dropbox/devs/ws/z-components/common/x_syslog.h:27,
                 from C:/Dropbox/devs/ws/z-components/log/include/esp_log.h:22,
                 from C:/Dropbox/devs/ws/z-sdk/esp-idf/components/bootloader_support/src/bootloader_random.c:22:
C:/Dropbox/devs/ws/z-sdk/esp-idf/components/esp32/include/esp_wifi.h:62:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
make: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/project.mk:450: component-common-build] Error 2
make[1]: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/component_wrapper.mk:274: app_trace.o] Error 1
make: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/project.mk:450: component-app_trace-build] Error 2
make[1]: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/component_wrapper.mk:274: bt.o] Error 1
make[1]: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/component_wrapper.mk:274: actuators.o] Error 1
make: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/project.mk:450: component-bt-build] Error 2
make[1]: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/component_wrapper.mk:273: src/bootloader_random.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/project.mk:450: component-actuators-build] Error 2
CC build/bootloader_support/src/flash_encrypt.o
CC build/coap/coap_debug.o
CC build/bootloader_support/src/bootloader_sha.o
CC build/coap/coap_observe.o
In file included from C:/Dropbox/devs/ws/z-components/hal_esp32/hal_network.h:7:0,
                 from C:/Dropbox/devs/ws/z-components/common/x_network.h:28,
                 from C:/Dropbox/devs/ws/z-components/common/x_syslog.h:27,
                 from C:/Dropbox/devs/ws/z-components/log/include/esp_log.h:22,
                 from C:/Dropbox/devs/ws/z-sdk/esp-idf/components/bootloader_support/src/flash_encrypt.c:25:
C:/Dropbox/devs/ws/z-sdk/esp-idf/components/esp32/include/esp_wifi.h:62:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.
make[1]: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/component_wrapper.mk:274: src/flash_encrypt.o] Error 1
make: *** [/c/Dropbox/devs/ws/z-sdk/esp-idf/make/project.mk:450: component-bootloader_support-build] Error 2
AR build/coap/libcoap.a
'

igrr, 2018-02-14:
In the components list:

```
/c/Dropbox/devs/ws/z-support/freertos
```

This component has the same name as ESP-IDF freertos component, so the same override mechanism prevents the one in IDF from being used. Hence errors related to missing FreeRTOS include files. Please rename z-support/freertos to something else.

ammaree, 2018-02-14:
Hi @igrr thanks for the feedback. It fixed the problem but something is a bit strange, I had a component.mk file in the 'z-support/FreeRTOS' directory with the following statements, based on the believe this would eliminate the whole directory.
COMPONENT_SRCDIRS		:=
COMPONENT_ADD_LDFLAGS	:=

Last question is based the output as intercepted from the ESP_LOG?() macros, is there any way to change what is currently hardcoded at compile time (using the macros) into a function call that can be intercepted and formatted in proper syslog format?
-----------------------------------------------------------------
 0.339: main wifi I (339) wifi: wifi firmware version: ebd3e5d

 0.339: main wifi I (339) wifi: config NVS flash: disabled

 0.339: main wifi I (339) wifi: config nano formating: disabled

 0.340: main wifi I (340) wifi: Init dynamic tx buffer num: 32

 0.340: main wifi I (340) wifi: Init data frame dynamic rx buffer num: 32

 0.341: main wifi I (341) wifi: Init management frame dynamic rx buffer num: 32

 0.341: main wifi I (341) wifi: wifi driver task: 3ffc3998, prio:23, stack:4096

 0.342: main wifi I (342) wifi: Init static rx buffer num: 10

 0.342: main wifi I (342) wifi: Init dynamic rx buffer num: 32

 0.343: main wifi I (343) wifi: wifi power manager task: 0x3ffc86ec prio: 21 stack: 2560

 0.414: wifi wifi I (414) wifi: mode : sta (30:ae:a4:32:bc:00)

 0.415: main app_main [start]
 0.416: TelNet vTaskTelnet [start]
 0.416: HTTP vHttpServerTask [start]
 0.416: Sensors vTaskSensors [start]
 0.416: Events vTaskEvents [start]
 0.417: MqttTx vMqttTaskTx [start]
 0.417: SNTP vSntpTask [start]
 0.417: Console vTaskConsole [start]
 0.417: Actuator vActuatorTask [start]
 0.535: wifi wifi I (535) wifi: n:1 1, o:1 0, ap:255 255, sta:1 1, prof:1

 1.525: wifi wifi I (1525) wifi: state: init -> auth (b0)

 1.527: wifi wifi I (1527) wifi: state: auth -> assoc (0)

 1.540: wifi wifi I (1540) wifi: state: assoc -> run (10)

 1.556: wifi wifi I (1556) wifi: connected with irmacos, channel 1

 2.847: MqttRx vMqttTaskRx [start]
 4.541: pmT wifi I (4540) wifi: pm start, type:0

 4.677: MqttRx vMqttSystemCommandGPB New registration

closed 2018-02-16 by ammaree

cnlohr, 2019-11-06:
Hoping to re-open this.  How do you disable IDF components using the new build system?  I can't seem to get the old build system to work with the ESP32-S2.

dobairoland, 2019-11-06:
Hi @cnlohr. The `make` based build system only supports esp32 target. This should be the error message if you try to set `IDF_TARGET=esp32s2beta` with `make`.

Questions of type "how to" is better to ask in the forums: https://esp32.com/viewforum.php?f=23 and if you identify some issue then please open a new issue here, on Github. 

In the meantime, please take a look at the documentation: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html, more precisely the `REQUIRED_IDF_TARGETS` option.

For example, you can see here https://github.com/espressif/esp-idf/blob/master/components/esp32/CMakeLists.txt the related parts:
```cmake
idf_build_get_property(target IDF_TARGET)
if(NOT "${target}" STREQUAL "esp32")
    return()
endif()
```
and 
```
idf_component_register(...
                        REQUIRED_IDF_TARGETS esp32)
```

cnlohr, 2019-11-06:
My vote is to continue support for the makefile-based approach considering the ease of debugging and progressing when doing large-scale changes (for instance in my case, I am making non-lwip builds on the ESP32S2, since our IP stack is significantly lower latency and improved experience on the ESP32).  I'm guessing that Makefile ship has sailed, though? :(

I tried making a forum post in the forum, however, all posting was timing out last night.  Additionally all google results for doing this sort of thing end up in this thread, and thus the answer should be posted where others will find based on google.

For modifying the component behavior I understand using `idf_component_register` in the CMakeLists.txt file per component, however, I couldn't get it working at the project level, I am struggling to more specifically call out at the top level trying to do these things on a per-project basis:
(1) Please do/do not compile code for component X.
(2) Please do/do not include -D. -I flags for component X.

In some cases we need headers but compiled code is provided by a local component.

Simply placing a folder by the appropriate name inside of a `components` subfolder then adding a `CMakeLists.txt` to it does not seem to have the correct override behavior always.  It works in some situations and not in others.

dobairoland, 2019-11-07:
@cnlohr I think if you provide some minimal working examples then developers in the forum will be able to help you. IDF contains a lot of special build cases which was possible to port to cmake. I'm confident that there are solutions for you as well.

danillogg, 2020-06-02:
I am on a similar troubleshooting.
My user name is ABC&(company name) as I tried to instal the ESP-idf it has crated a path to a folder named just ABC now all Cmake actions are leading to a folder non existent.
How can I fix the pathway to this folder inside the ESP-idf installer?

dobairoland, 2020-06-03:
Hi @danillogg. Your issue doesn't seem to be similar to the above ones. Please open a new issue for it and while doing it please also provide the additional information suggested by the issue template. Thanks!