espressif/esp-idf #1617: AUTH_FAIL loop following disconnect until ESP32 is power cycled.

lonerzzz, 2018-02-12:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			2017/11/26
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Not sure if this issue already exists in the ESP32-IDF issues, but didn't see it. The issue appeared following update e5b2c1c  and remains until this at least 1c3dd23. The issue was originally raised under as (https://github.com/espressif/arduino-esp32/issues/918) but has been brought here because it is IDF related.

After one of the recent core updates, I have noticed that in the event that the WiFi gets disconnected or the application calls WiFi.disconnect(true|false) such as when a WiFi.scan() has to be called (due to a known bug https://github.com/espressif/arduino-esp32/issues/758), that it is often no longer possible to reconnect to the WiFi router until the ESP32 is power cycled. The problem happens about 1 in 3 times. Sometimes a simple reboot fixes the problem, but generally the power cycle is required.

Every attempt to make a connection with WiFi.begin results in AUTH_FAIL in this scenario despite the WiFi.begin parameters being correct:

[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL

The following message sometimes occurs as well, but not always so it is not clear :
[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 8 - ASSOC_LEAVE

CelliesProjects, 2018-02-17:
The issue is also still present in Arduino IDE, see (espressif/arduino-esp32#1001) and (espressif/arduino-esp32#674).

lonerzzz, 2018-02-17:
@CelliesProjects Can you add more detail regarding how the issues you referenced are related to the stated problem? Otherwise, it makes more sense to raise a separate issues so that we one solved but the other remaining because someone thinks they are the same thing.

CelliesProjects, 2018-02-17:
@lonerzzz 

> that it is often no longer possible to reconnect to the WiFi router until the ESP32 is power cycled. The problem happens about 1 in 3 times. Sometimes a simple reboot fixes the problem, but generally the power cycle is required.

This, and problems with SmartConfig, which has worked pretty good until october 2017 which isn't that far from e5b2c1cf4c08471c740d3f4e5f86012aa1f32894 (23 Nov 2017).

tferrin, 2018-02-27:
I am experiencing a similar problem that appeared after updating to a newer arduino-esp32 repo in late December. Prior to that my WiFi connections were always quite reliable, but since then I get many DISCONNECT events caused by AUTH_FAIL's. Sometimes power cycling improves the situation, but not reliably so. If I move my ESP32 to within a few feet of my AP the rate of DISCONNECTs goes down, but never to zero, whereas previously I got great WiFi coverage. After banging my head on the table for what seems like days without finding any reliable solution, I ran across this issue that rings pretty true to me: https://github.com/espressif/esp-idf/issues/1366. I can't be sure now that so much time has passed, but I think my problems started after upgrading to the version of the repository that had the backwards-incompatible change to SPIFFS_OBJ_META_LEN. I'm thinking that somehow that change, perhaps in combination with running the old mkspiffs.exe tool, corrupted the WiFi RF calibration data that is stored in NVS. I think the SmartConfig-stopped-working issue noted above could also be related to RF calibration data corruption. If anyone knows how to force PHY_RF_CAL_FULL (as described in #1366) after a reboot when using the arduino-esp32 it would make for easy confirmation that corrupted calibration data is  the root cause of many of the reported DISCONNECT/AUTH_FAIL problems.

liuzfesp, 2018-03-05:
HI @tferrin, you can enable force RF calibration via the menuconfig:  make menuconfig -> component config -> PHY -> Disable "Store phy calibration data in NVS"

closed 2018-04-04 by FayeY

tferrin, 2018-04-05:
I had lost track of this particular WiFi issue thread and should have followed up earlier. There was another related thread (https://github.com/espressif/esp-idf/issues/1366) discussing potential corruption of NVS, benefits of doing a full RF calibration on every restart, etc. As I posted in that thread, in my case I wrote some check-summing code to verify that NVS is not getting corrupted and that doing a full RF calibration makes no difference to the DISCONNECT/AUTH_FAIL problem. In the mean time have determined that the problem is worse with some brands of WiFi access points than others, with Cisco APs particularly problematic, but found nothing specific to point a finger at. As noted in the thread referenced above, v3.1 of the IDF will provide more WiFi debugging options so hopefully we can all get to the bottom of all of these seemingly related WiFi problems soon.

It is frustrating that there are many threads reporting DISCONNECT/AUTH_FAIL WIFI problems but as far as I can tell they are not linked to each other anywhere. This results in many duplicate reports and no single source of information, which results in lots of wasted time searching through similar issue reports. I don't have any brilliant solution to propose for this, but I wish I had my wasted time back.