espressif/esp-idf #1619: UART1 as CONFIG_CONSOLE_UART_NUM issue

hvishu, 2018-02-13:
If we enable UART1, (Tx-pin10, Rx-pin9) as UART for console output using menuconfig, get a compilation error in /components/esp32/clk.c. (line number 198, "DPORT_UART0_CLK_EN", not defined).  Attached is sdkconfig file.
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/1719098/sdkconfig.zip)

Seems the below logic is not correct... 
#if CONFIG_CONSOLE_UART_NUM != 0
                              DPORT_UART0_CLK_EN |
#endif
#if CONFIG_CONSOLE_UART_NUM != 1
                              DPORT_UART1_CLK_EN |
#endif
#if CONFIG_CONSOLE_UART_NUM != 2
                              DPORT_UART2_CLK_EN |
#endif

For time being I have made it work by disabling DPORT_UART0_CLK_EN , but somehow it still works for me.  I am using git version "e4dd0b498bfeebcecb3ff77999f30536f8091250".  Not sure whether this has been already fixed in the latest version

igrr, 2018-02-13:
This has been fixed in
https://github.com/espressif/esp-idf/commit/a70522149297b4b254e3bfb43340346ef167114b

hvishu, 2018-02-13:
Ok. I see the definition of  DPORT_UART0_CLK_EN in dport_reg.h.  I will not be able to test this version immediately, so closing the issue.

closed 2018-02-13 by hvishu