espressif/esp-idf #1621: BT_BLE_DYNAMIC_ENV_MEMORY fails to compile when not enabling Gatt Client

spli, 2018-02-13:
I've been trying the new BT_BLE_DYNAMIC_ENV_MEMORY option as I want to control when BLE memory is allocated. My project just uses Gatt Server module, so GATTC_ENABLE=n.

When compiling I get the following error:

`build/bt/libbt.a(bte_init.o):(.literal.BTE_InitStack+0x0): undefined reference to `bta_gattc_cb_ptr'`

I can only get it to compile with GATTC_ENABLE=y. Which I don't want to do as I'm not using the client.

This is on master.

Thanks

spli, 2018-02-14:
I've also noticed that it doesn't seem to do anything, memory doesn't seem to by dynamically allocated on use. Or maybe I've misunderstood this feature?

Weijian-Espressif, 2018-02-28:
@Yulong-espressif please help to view the question.

Weijian-Espressif, 2018-02-28:
@spli sorry, it is a realy bug. we will fix it at the latest IDF. 
You can delete line 272 "memset((void *)bta_gattc_cb_ptr, 0, sizeof(tBTA_GATTC_CB));" in void BTE_InitStack(void) of components/bt/bluedroid/main/bte_init.c.

FayeY, 2018-03-05:
Hi @spli , this issue should have been solved in the latest IDF, please have a try and feel free to reopen it if the issue persists. Thanks.

closed 2018-03-05 by FayeY