espressif/esp-idf #1622: compile_only_if doesn't work

mfp20, 2018-02-13:
component.mk

```
COMPONENT_SRCDIRS:=sources/lib sources/lib/vauth sources/lib/vtls
COMPONENT_PRIV_INCLUDEDIRS:=sources/lib sources/include
CFLAGS+=-DHAVE_CONFIG_H -DBUILDING_LIBCURL

$(call compile_only_if,$(CONFIG_COMPONENT_CURL),*.o)
```

sdkconfig

```
COMPONENT_SRCDIRS:=sources/lib sources/lib/vauth sources/lib/vtls
COMPONENT_PRIV_INCLUDEDIRS:=sources/lib sources/include
CFLAGS+=-DHAVE_CONFIG_H -DBUILDING_LIBCURL

$(call compile_only_if,$(CONFIG_COMPONENT_CURL),*.o)
```

make all

```In file included from components/curl/sources/lib/splay.c:23:0:
components/curl/sources/lib/curl_setup.h:56:25: fatal error: curl_config.h: No such file or directory
compilation terminated.```

fri-sch, 2018-07-22:
It seems like it still doesn't work. Components are compiled irrespective of the config option.

fri-sch, 2018-11-03:
I tested with v3.1.1 and it seems to work. So I guess this can be close.

closed 2018-11-03 by Alvin1Zhang