espressif/esp-idf #1625: Poor WiFi AP performance on Rev 1 chips on specific channels

mill1000, 2018-02-14:
I am observing poor AP performance on Rev 1 chips, when the channel is set to 2, 4 or 5. The same code operates well on Rev 0 chips. The WiFi environment on those channels is relatively free from interference.

Wireshark shows occasional TCP re-transmissions, but additional testing showed TCP and UDP to be equally effected.

Currently building with esp-idf commit cf2600a

negativekelvin, 2018-02-14:
Did you try to erase flash or force full calibration?

mill1000, 2018-02-14:
I have performed make erase_flash. Is there another method of triggering calibration?

negativekelvin, 2018-02-15:
There is a menuconfig option to not store calibration so it will be performed every boot

mill1000, 2018-02-15:
Additional testing has found that this problem only occurs in some of my modules. Is it likely a result of their installation, and not an underlying issue with the Rev 1 chip.

Apologies, I jumped the gun on this one.

closed 2018-02-15 by mill1000