espressif/esp-idf #1628: ble esp_ble_gattc_register_for_indicate

chegewara, 2018-02-16:
Hi, is there a plan to add function to register for indications similar to `esp_ble_gattc_register_for_notify`. I know i can register for indications by writing 0x0200 to proper descriptor, but it would be nice to have API to do it.

heyinling, 2018-02-23:
this API is used for both notify and indication. As I know calling this API won't auto write for CCC. It just allows BT stack to send event to app when receive notify/indicate.

chegewara, 2018-02-23:
@heyinling You are saying about client event that is used when indication or notification is received which is described by this structure
http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gattc.html?#_CPPv2N24esp_ble_gattc_cb_param_t22gattc_notify_evt_paramE 
and bool param is_notify determines if this is notification or indication, but im talking about this function
http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gattc.html?#_CPPv233esp_ble_gattc_register_for_notify13esp_gatt_if_t13esp_bd_addr_t8uint16_t
which is writing to CCC value 0x0100, but to register for indications we need to write 0x0200 to CCC and such is required new API.

heyinling, 2018-02-23:
@chegewara I mean `esp_ble_gattc_register_for_notify` won't write CCC. it can be used for both notification and indication. But I haven't tested it for some time, not 100% sure about the current behavior.

could you please help to check if calling `esp_ble_gattc_register_for_notify` in client side will write CCC? you may use 2 ESP32 board and check if GATTS side receive CCC write event.

chegewara, 2018-02-23:
Im pretty sure esp_ble_gattc_register_for_notify does write CCC, or at least its how i understand how it works from this library
https://github.com/nkolban/esp32-snippets. Its enough to register for notify and it works, there is no need to write to CCC
https://github.com/nkolban/esp32-snippets/blob/56790e1b1b05fb0f4c1944ff49e63a11e55d08b8/cpp_utils/BLERemoteCharacteristic.cpp#L486

heyinling, 2018-02-24:
@chegewara I've checked with latest master and I'm sure `esp_ble_gattc_register_for_notify ` will not write CCC. writing CCC is not always mandatory (depending on BT stack implementation) for client to recv notify/indicate, although we suggest to do that for best compatibility .

you may have a try of the following cases:

case 1, should receive indicate:

1. call `esp_ble_gattc_register_for_notify ` for indicate char
2. don't write to CCC by your app code
3. send indicate form GATTS

case 2, should not receive indicate:
1. don't call `esp_ble_gattc_register_for_notify ` for indicate char
2. write to CCC by your app code
3. send indicate from GATTS

the 2 cases above suggested that `esp_ble_gattc_register_for_notify` is only a local function for GATTC, used to control if BT stack need to notify app when it receives notify/indicate.

chegewara, 2018-02-24:
@heyinling Ok, so its only internal function to register internaly (client side) to receive ESP_GATTC_NOTIFY_EVT for both, indications and notifications. Thanks.

closed 2018-02-24 by chegewara