espressif/esp-idf #1629: uart_set_sw_flow_ctrl don't allow large thresholds numbers

copercini, 2018-02-16:
Hi!

I'm working on a UART software flow control with Xon and Xoff
My UART buffer has 4096 bytes, so I'm planning to set Xon threshold as 1024 and Xoff as 3072 but the ESP32 API only allows uint8_t (255 bytes) thresholds here: https://github.com/espressif/esp-idf/blob/master/components/driver/uart.c#L228

Am I being overstated with thresholds or is uint8_t too low for this API?

negativekelvin, 2018-02-17:
Well the threshold applies to the fifo data not the buffer and the limit is 1024 bytes (default 128) so the threshold limit is 10 bits (8 as implemented).  To get larger thresholds you will have to dynamically enable or disable it based on the amount of data in the buffer.

FayeY, 2018-02-23:
Hi @copercini , did the above suggestion help you resolve the issue?

copercini, 2018-02-23:
Yes, thanks @negativekelvin

closed 2018-02-23 by copercini