espressif/esp-idf #1630: Error -76 from MbedTLS

ammaree, 2018-02-16:
I am getting a -76 error code (undefined) from MbedTLS when accessing a Google API

the symptom is exactly the same @nkolban  described in
[https://github.com/curl/curl/issues/1327](url)

I have not been able to get any other references to the same problem, has this been addressed?

Thanks

Andre

projectgus, 2018-02-18:
Hi,

Thanks for reporting this Andre. This looks like a duplicate of #434. That issue seems to have fallen through the cracks with regards to being fixed, so thanks for poking us about it.

I'm going to close this as a duplicate, but please click on the linked issue and Subscribe to notifications there.

closed 2018-02-18 by projectgus

projectgus, 2018-02-19:
Hi Andre,

Sorry I didn't look into this properly this morning. The linked issue was actually fixed some time ago, but the Github issue had not been closed.

Result -76 is -0x004C which is `MBEDTLS_ERR_NET_RECV_FAILED`.  This return value essentially means that the underlying socket read failed for some reason, maybe to do with the network or because the other end hung up unexpectedly. It means you need to reset the SSL session and start again with a new socket.

If you think ESP-IDF is doing the wrong thing here then you could try logging the value of `error` at this point:
https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/net_sockets.c#L368

Please reopen if you think there's another issue here to look into from the IDF side.

cipibad, 2019-03-06:
Hi,

I reproduced similar problem. I have one mqtts connection and a second https connection for ota.
ota file transfers fails randomly with `esp-tls: read error :-76:`.
The underlying socket error is 11(`EAGAIN`)

Thank you,
C

bearsh, 2019-05-21:
I also encounter this issue. it seems to be wrongly set if the underlying socket would return `EAGAIN`. If I do a `esp_tls_conn_read()` in a loop, it gets returned every `esp_tls_cfg_t.timeout_ms` if nothing is received. the connection just works fine in spite of the error...

milamber-ls, 2019-06-25:
I have this issue as well