espressif/esp-idf #1632: check-lxdialog.sh on latest Msys2/mingw - can't find ncurses.

greblus, 2018-02-16:
Hi guys.

I've been trying to set-up esp-idf on a freshly updated Msys2 installation on Windows 7 and  it turned out that tools/kconfig/lxdialog/check-lxdialog.sh script was unable to find ncurses and linker flags so make menuconfig was failing.

To make it work I've temporarly fixed it in this way:

# What library to link
ldflags()
{
   echo "-lncurses -L/usr/lib/"
}

# Where is ncurses.h?
ccflags()
{
   echo '-I/usr/include/ -DCURSES_LOC="<curses.h>"'
}

Unfortunately I wasn't able to fix the script yet but I'll keep trying :). I attach my ad-hoc solution which works for me.

[check-lxdialog_sh.txt](https://github.com/espressif/esp-idf/files/1731607/check-lxdialog_sh.txt)

wayneuroda, 2018-02-27:
After I made (and committed) this change, I had to forcibly clean my git repo (using `git clean -xdf`) in order to get the build to work, otherwise I was getting errors about missing references in libintl, even `make clean` didn't seem to actually clean anything. Maybe this will help somebody else not stumble around on this like I did!

greblus, 2018-02-27:
Unfortunately it didn't help in my case and the script still can't find ncurses.

FHFS, 2018-03-09:
I remember some kind of issue with line endings in the check-lxdialog.sh file. Try CR LF, and just LF as line endings.

greblus, 2018-03-20:
I think that the problem is in version of gcc which is being used to compile tests run by check-lxdialog.sh. 
In tools/kconfig/Makefile where check-lxdialog.sh is invoked $CC is set to /usr/bin/gcc while when using mingw32.exe it should be /mingw32/bin/gcc.

greblus, 2018-03-21:
Ok, finally I discovered this bug report:

https://github.com/espressif/esp-idf/issues/1296

and kconfig must be compiled on Windows using MSYS toolchain, so I think the only way to fix it is to add something like:

    if [ -z $(uname -s)##*"MINGW32"* ]; then
       echo "-lncurses -L/usr/lib/"
       exit 1
    fi

on top of ldflags() and

    if [ -z $(uname -s)##*"MINGW32"* ]; then
        echo '-I/usr/include/ -DCURSES_LOC="<curses.h>"'
       exit 1
    fi

on top of ccflags().

I attach my modified check-lxdialog.sh.
[check-lxdialog.sh.txt](https://github.com/espressif/esp-idf/files/1832448/check-lxdialog.sh.txt)

cmbahadir, 2019-05-24:
Hello,

I suffered this problem with latest build (v3.2) and solved it by applying the change of @greblus , but i also had to change "/usr/lib" as "/mingw32/lib" and "/usr/include" as "/mingw32/include/ncurses".

So, is it intentionally left open or is there a reason to not create a pull request and merge?

Have a nice weekend :) 
Mete

dobairoland, 2022-04-08:
The Mingw/MSYS was deprecated in ESP-IDF v4.0 and removed entirely in v5.0. 

The same goes for the menuconfig library. It is based on https://pypi.org/project/kconfiglib/ so this issue cannot happen anymore.

closed 2022-04-08 by dobairoland