espressif/esp-idf #1635: Modem Sleep corrupts UART input

jamierobertson1, 2018-02-18:
When enabling modem sleep with the current master, UART (in my case UART2) received data is garbled.

I just set `esp_wifi_set_ps(WIFI_PS_MODEM);`  after '(esp_wifi_start())'. This worked ok with version 2.1.1.

If I don't enable modem sleep the received data is fine.

panfeng-espressif, 2018-02-28:
Hello jamierobertson1,
What's wrong with your uart data? Could you please send the log for analyzing?
It works fine for IDF version 2.1.1 and gets error for the latest IDF, is that correct?

jamierobertson1, 2018-02-28:
Yes, that's correct. I'm currently away for the next few days, but will send a log when I get back.

All I'm doing is displaying the results from `uart_read_bytes` using ESP_LOG. As soon as esp_wifi_set_ps(WIFI_PS_MODEM) is enabled, instead of displaying the received characters I'd expect, I get random characters like spaces ,ampersands and question marks.

jamierobertson1, 2018-03-03:
You can close this. It now works with the latest master. I'm not sure what the problem or fix was.

closed 2018-03-03 by jamierobertson1