espressif/esp-idf #1636: Strcpy-ing SSID and Password into wifi_config.sta.ssid and wifi_config.sta.password renders the WiFi remains disconnected to AP

uncle-yong, 2018-02-18:
Hi there,

ESP-IDF version - latest.

Example program: http_request

This http_request when compiled in C and flashed into the ESP32, it works as usual.

Now I'm trying to compile it in C++ so that I can use the iomanip and sstream for some JSON parsing. However, it failed to compile, saying that "sorry, unimplemented: non-trivial designated initializers not supported" on that declaration:

main.c:

> #define EXAMPLE_WIFI_SSID "network1"
> #define EXAMPLE_WIFI_PASS "abcd123"
> ... more code
> wifi_config_t wifi_config = {
        .sta = {
            .ssid =     EXAMPLE_WIFI_SSID,
            .password = EXAMPLE_WIFI_PASS,
        },
    };

So after that, I changed it to this:

main.cpp:

> const char EXAMPLE_WIFI_SSID[] =  "network1";
> const char EXAMPLE_WIFI_PASS[] =  "abcd123";
> ... more code
>   wifi_config_t wifi_config;
>   strcpy((char*)wifi_config.sta.ssid, (const char*)EXAMPLE_WIFI_SSID);
>   strcpy((char*)wifi_config.sta.password, (const char*)EXAMPLE_WIFI_PASS);`

[source: https://www.esp32.com/viewtopic.php?t=1317]

It compiles without a problem. But, it just doesn't connect to AP. The state is always _SYSTEM_EVENT_STA_DISCONNECTED_ no matter what. The password and the SSID are all correct.

Any ideas what is going on? Have I missed out something there?

mahavirj, 2018-02-18:
@uncle-yong Could you please check if zero initializing (or making static) wifi_config struct changes the behavior or not?

uncle-yong, 2018-02-19:
Hello,

Thanks for the tip, it works now, however, there are warnings like:

> D:/esp32-projects/http_request/main/main.cpp: In function 'void initialise_wifi()':
D:/esp32-projects/http_request/main/main.cpp:133:42: warning: missing initializer for member 'wifi_ap_config_t::password' [-Wmissing-field-initializers]
     static wifi_config_t wifi_config = {0};
                                          ^
D:/esp32-projects/http_request/main/main.cpp:133:42: warning: missing initializer for member 'wifi_ap_config_t::ssid_len' [-Wmissing-field-initializers]
D:/esp32-projects/http_request/main/main.cpp:133:42: warning: missing initializer for member 'wifi_ap_config_t::channel' [-Wmissing-field-initializers]
D:/esp32-projects/http_request/main/main.cpp:133:42: warning: missing initializer for member 'wifi_ap_config_t::authmode' [-Wmissing-field-initializers]
D:/esp32-projects/http_request/main/main.cpp:133:42: warning: missing initializer for member 'wifi_ap_config_t::ssid_hidden' [-Wmissing-field-initializers]
D:/esp32-projects/http_request/main/main.cpp:133:42: warning: missing initializer for member 'wifi_ap_config_t::max_connection' [-Wmissing-field-initializers]
D:/esp32-projects/http_request/main/main.cpp:133:42: warning: missing initializer for member 'wifi_ap_config_t::beacon_interval' [-Wmissing-field-initializers]

Filling them with zeros doesn't help as these warnings are still there after compiling.

There is a similar issue to this one after I read the last message - if your team can locate that, maybe you can close this as "Duplicate". Thanks for the help too.

projectgus, 2018-02-19:
Hi @uncle-yong ,

To fix this warning, remove the first zero from between the initializer braces ie

```
static wifi_config_t wifi_config = { };
```

This will initialize the structure to all-zeroes (rather than uninitialized memory), without the C++ compiler wanting initializers for all elements.

uncle-yong, 2018-02-19:
Hello, thanks for the suggestion. These warnings are wiped out and worked well in the application.

Alright to suggest if these examples are written so that it's portable to both C and C++ ?

projectgus, 2018-02-19:
> Alright to suggest if these examples are written so that it's portable to both C and C++ ?

Suggestions are always welcome. In general, we try to make sure the C APIs we provide work with C++.

C99 named initializers are a big pain here because they're the most obvious way in which g++ does not compile a superset of standard C. However, named initializers are also really good for code readability - if you compare the original named initializer version with the `strcpy()` version then the first one is easier to read and understand and less prone to accidental errors. (Even though the compiler will optimize and produce the same binary in both cases, probably.)

The best advice I can offer is to either put the C API interfacing code you have into a separate .c file and expose an `extern "C"` function interface which you can call from your C++ code, or look into the (third-party) C++ wrapper API efforts from Neil Kolban in his esp32-snippets:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/WiFi.h

uncle-yong, 2018-02-19:
Thanks for your help. I'm closing this issue as the program works with that suggestion.

closed 2018-02-19 by uncle-yong

tanmaysathe, 2020-04-04:
Setting initial values to 0 works.
memset(&wifi_config, 0, sizeof(wifi_config));