espressif/esp-idf #1637: Shouldn't esp_ping_set_target() be using timeout values of milliseconds?

chmorgan, 2018-02-18:
esp_ping_set_target() is multiplying the incoming value by 1000, so if you pass in milliseconds you end up with seconds per the usage of sys_msleep() and other functions in ping.c

```
    case PING_TARGET_RCV_TIMEO:
        ESP_PING_CHECK_OPTLEN(opt_len, uint32_t);
        ping_option_info->ping_rcv_timeout = (*(uint32_t *)opt_val) * 1000;
        break;
    case PING_TARGET_DELAY_TIME:
        ESP_PING_CHECK_OPTLEN(opt_len, uint32_t);
        ping_option_info->ping_delay = (*(uint32_t *)opt_val) * 1000;
```

The ping thread in ping.c uses sys_msleep() which is documented as:

```
/* Time functions. */
#ifndef sys_msleep
void sys_msleep(u32_t ms); /* only has a (close to) 1 ms resolution. */
#endif
```

It looks like those *1000 should be removed (and it would probably be helpful to document the units for the delay and timeout parameters). I can submit a PR if you guys agree with the change.

chmorgan, 2018-02-18:
Pushed to https://github.com/espressif/esp-idf/pull/1638

igrr, 2018-03-20:
Referenced PR has been merged, closing the issue.

closed 2018-03-20 by igrr