espressif/esp-idf #1639: Poor TCP performance when BLE scan

hopkinskong, 2018-02-19:
I have a single TCP connection to a server (constant, stay connected forever), there is a task pushing data to the TCP socket every 200ms, when the connection stays idle for more than 6.4s (6400ms), the server will close the TCP connection (a custom timeout mechanism).

My ESP32 will keep scanning BLE devices (BLE only, no classic Bluetooth), passive scan, every 1s, forever. The issue is when BLE is keep scanning BLE devices, my TCP connection will disconnect for a long time (which means there is at least 6.4s for no data in the TCP socket stream), some of the data is delayed for few seconds to arrive to my server too.

Is there any way/tweak that I could reliably use Wi-Fi+BLE? 

I am using ESP32-WROVER if that matters.

jedi7, 2018-02-19:
Hi,
I'm using also BLE scan + Wifi, but I do not see too big slow down. For example my BLE scans and also I'm able to do OTA (I can see a 1s delay during OTA write and BLE scan, but  then it continue properly). 

Anyway be aware ESP have common RF part for Wifi and Bluetooth, so it is switching between them.
But maybe a devs will tell more :)

hopkinskong, 2018-02-19:
@jedi7 Thanks. I think the problem is I have a constant TCP connection to my server, the BLE scan will delayed my TCP data stream. Maybe the issue is not exists when you open a TCP connection every time you want to send a data out.

hopkinskong, 2018-03-05:
Cross reference back to #1658

TianHao-Espressif, 2018-04-03:
@hopkinskong , you can try the latest master branch. We have done the wifi/bluetooth coexistence.

hopkinskong, 2018-04-03:
@TianHao-Espressif Replied in #1658

closed 2018-08-21 by FayeY