espressif/esp-idf #1640: Changing the partition table (only the coredump) using API

ammaree, 2018-02-19:
We have a number of device deployed in the field with the all the usual NVS and OTA data partitions, followed by 3x OTA partitions of equal size.

On the first device deployed we created a 64K "core dump" partition immediately following the #3 OTA partition, with the last 700K or so free.

Since we have completely changed our approach to the configuration and use of core dump we would now like to change the size (and possibly location) of the "core dump" partition programmatically, without having to physically drive to, open and completely reflash each device.

Is this possible, and if so how?

Thanks

ammaree, 2018-02-19:
Have had another look at the existing APIs and in theory I am sure it can be done by
a) physically reading the partition table at 0x8000,
b) working through the structure to find the coredump partition entry
c) verifying the details to make sure we have the required spare space both in size and location
d) possibly (optional) to wipe the existing coredump partition space
e) modifying the coredump entry, mainly address and size in the partition table
f) writing the modified partition table back to flash.

Now whilst the above theory might sound simple and straightforward, the practice of implementing this in the light of the complexity of memory mapping and the inter/locking of CPU activity makes it a very risky and complex issue, something I would at this stage prefer to avoid doing myself.
 
Any chance that this is on the tasklist for somebody, I think it will we of huge value for many people to enable in the field FOTA upgrades on production devices...

All feedback appreciated.

projectgus, 2018-02-19:
Hi @ammaree,

Unfortunately, we don't support OTA updating of partition tables. Because the entire 4KB flash sector needs to be erased and then rewritten, there is a short window when a power failure or crash would result in a bricked device requiring physical reflashing via serial. If possible, find a way to work with the legacy partition table you have.

However, if you're sure that a power failure won't occur (and you're not moving at least one of your app partitions) and you're prepared to be very careful then you can rewrite the partition table in place.

*Please note that this is entirely unsupported and at your own risk*:

1. Erase sector at 0x8000.
2. Write new partition binary data at 0x8000. You can take the new partition .bin file from the build directory of your app, and follow these steps to embed it as [accessible binary data inside the app itself](https://esp-idf.readthedocs.io/en/latest/api-guides/build-system.html#embedding-binary-data).
3. Reset and let the ESP32 can boot from the new configuration.

Note that any power failure or crash between step 1 & the end of step 2 will probably result in an unbootable device. This window is thankfully only a few tens of milliseconds wide, the exact timing will depending on your flash chip.

Suggest having your app do this directly from app_main, before initialising WiFi or any other tasks, to minimise the chance of unexpected problems. The app can compare the existing partition table with the embedded one to determine if it needs to update or not.

Note that changing the partition table can change the offset of one or more other app or data partitions. Your ESP32 will need a working app to boot from after reset - this will be determined by the ota_data partition (if it's moved then you'll need to rewrite it) and the offsets of the app partitions. If you can avoid moving these, then that's by far the easiest approach.

Also note that recent IDF (v3.1-dev) now appends an md5sum to the partition table, and older bootloaders will fail to boot with these partition tables. A backwards compatibility option in menuconfig (to disable the md5sum) should be merged to master this week.

ammaree, 2018-02-21:
Hi @projectgus 

Thanks for the feedback. For now we are ONLY looking at changing the coredump and want to move it to the end of flash. This will leave the section between 0x00310000 and 0x003CFFFF free for later use.

We would like to use this space for a flash file system, but concerned about future OTA application size needs, hence for now want to leave the space free. This will allow us to expand the OTA partition(s) in future.

Your suggestion to do these changes very early makes sense although our devices are all powered off a large 24V 1000Ah DC battery platform. The result of corrupting the partition table is same as not implementing the functionality, having to go to each site and physically reflash each device, only difference would be that if we do not implement the functionality, the visit to site would be scheduled proactively with less downtime
 
Q1: We would prefer to not include a full 3K in the application if we can rather build the logic to read, change in ram, erase and write the updated partition back. To do this we would need to recalc an MD5 checksum and rewrite that as well?

Q2: The coming change to disable the MD5 checksum would remove the need to recalc the checksum, but would reduce data integrity checking?

ammaree, 2018-02-24:
@projectgus 

Thanks for the option to disable the MD5 checksum. Is there any chance for some pointers on manipulating the core dump partition entry directly, structure/rules etc, to avoid sending a whole partition table down?

I also have a separate issue open regarding the size of the actual core dump, any input on that will also be appreciated...

Thanks

closed 2018-02-25 by projectgus