espressif/esp-idf #1642: NVS Blob and String operations leak memory

tommybazar, 2018-02-19:
After spending some time looking for a slow, but persistent memory leakage problem in our code, I found that nvs_set_blob allocates some overhead memory that is not freed on nvs_erase_key, nor on nvs_erase_all nor on handle close. Other nvs_setXXX functions also allocate this memory, but erasing the given key does free it. 

nvs_set_str is does something similar, but it does free about half of the memory it allocates...

Below is a small program demonstrating this.

We're using release v3.0 from January 16th (commit e7dc749). None of the commits from last month seem to have touched the NVS.

```
#include <iostream>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <nvs.h>
#include <sstream>
#include "esp_system.h"
#include <esp_log.h>
#include <nvs_flash.h>

#define MAIN_TAG "main"

std::string makeFileName(uint32_t index) {
  std::ostringstream outStr;
  outStr << "testname_" << index;
  return outStr.str();
}

extern "C" void app_main() {

  uint8_t buffer[255] = {0xFF};
  uint8_t buffer2[255] = {0x00};

  nvs_flash_init();

  esp_err_t error = 0;
  nvs_handle nvsHandle;
  nvs_open("test2", NVS_READWRITE, &nvsHandle);

  // Set blobs
  for (int i = 0; i < 100; i++) {
    error = nvs_set_blob(nvsHandle, makeFileName(i).c_str(), buffer, 255);
    nvs_commit(nvsHandle);
    ESP_LOGI(MAIN_TAG, "set blob @%d, error = %d, heap = %d", i, error, esp_get_free_heap_size());
    usleep(10);
  }

  // Get blobs to prove they're really there
  for (int i = 0; i < 100; i++) {
    uint32_t length = 256;
    error = nvs_get_blob(nvsHandle, makeFileName(i).c_str(), buffer2, &length);
    nvs_commit(nvsHandle);
    ESP_LOGI(MAIN_TAG, "get blob @%d, error = %d, heap = %d", i, error, esp_get_free_heap_size());
    usleep(10);
  }

  // Erase blobs to see them not giving back memory
  for (int i = 0; i < 100; i++) {
    error = nvs_erase_key(nvsHandle, makeFileName(i).c_str());
    nvs_commit(nvsHandle);
    ESP_LOGI(MAIN_TAG, "removed blob @%d, error = %d, heap = %d", i, error, esp_get_free_heap_size());
    usleep(10);
  }

  // Get blobs with error to prove they're really NOT there
  for (int i = 0; i < 100; i++) {
    uint32_t length = 0;
    error = nvs_get_blob(nvsHandle, makeFileName(i).c_str(), buffer2, &length);
    nvs_commit(nvsHandle);
    ESP_LOGI(MAIN_TAG, "get blob @%d, error = %d, heap = %d", i, error, esp_get_free_heap_size());
    usleep(10);
  }

  nvs_erase_all(nvsHandle);
  ESP_LOGI(MAIN_TAG, "2 Heap after delete all %d", esp_get_free_heap_size());
  nvs_close(nvsHandle);
  ESP_LOGI(MAIN_TAG, "2 Heap after close %d", esp_get_free_heap_size());
  nvs_open("test2", NVS_READWRITE, &nvsHandle);
  ESP_LOGI(MAIN_TAG, "2 Heap after open %d", esp_get_free_heap_size());

  // Add ints to see them consume heap.
  for (int i = 0; i < 100; i++) {
    error = nvs_set_i32(nvsHandle, makeFileName(i).c_str(), 1);
    nvs_commit(nvsHandle);
    ESP_LOGI(MAIN_TAG, "set int @%d, error = %d, heap = %d", i, error, esp_get_free_heap_size());
    usleep(10);
  }

  // Remove them to see them give it back.
  for (int i = 0; i < 100; i++) {
    error = nvs_erase_key(nvsHandle, makeFileName(i).c_str());
    nvs_commit(nvsHandle);
    ESP_LOGI(MAIN_TAG, "removed int @%d, error = %d, heap = %d", i, error, esp_get_free_heap_size());
    usleep(10);
  }

  nvs_close(nvsHandle);
  usleep(10000000);
}

```

Cheers,
Tomas.

negativekelvin, 2018-02-19:
Can you post the resulting output of your ESP_LOGI calls for reference

agmuntianu, 2018-02-20:
Because I am also hunding a memory leak somewhere I ran @tommybazar code. My IDF is 8856cc055a7d2b0 , this is the output:

<details>
  <summary>Long Logs</summary>
<pre>
I (533) main: set blob @0, error = 0, heap = 294476
I (533) main: set blob @1, error = 0, heap = 294308
I (533) main: set blob @2, error = 0, heap = 294308
I (543) main: set blob @3, error = 0, heap = 294308
I (543) main: set blob @4, error = 0, heap = 294308
I (553) main: set blob @5, error = 0, heap = 294308
I (553) main: set blob @6, error = 0, heap = 294308
I (563) main: set blob @7, error = 0, heap = 294308
I (563) main: set blob @8, error = 0, heap = 294308
I (573) main: set blob @9, error = 0, heap = 294308
I (583) main: set blob @10, error = 0, heap = 294308
I (583) main: set blob @11, error = 0, heap = 294308
I (593) main: set blob @12, error = 0, heap = 294308
I (593) main: set blob @13, error = 0, heap = 294176
I (603) main: set blob @14, error = 0, heap = 294176
I (603) main: set blob @15, error = 0, heap = 294176
I (613) main: set blob @16, error = 0, heap = 294176
I (623) main: set blob @17, error = 0, heap = 294176
I (623) main: set blob @18, error = 0, heap = 294176
I (633) main: set blob @19, error = 0, heap = 294176
I (633) main: set blob @20, error = 0, heap = 294176
I (643) main: set blob @21, error = 0, heap = 294176
I (643) main: set blob @22, error = 0, heap = 294176
I (653) main: set blob @23, error = 0, heap = 294176
I (653) main: set blob @24, error = 0, heap = 294176
I (663) main: set blob @25, error = 0, heap = 294176
I (673) main: set blob @26, error = 0, heap = 294176
I (673) main: set blob @27, error = 0, heap = 294044
I (683) main: set blob @28, error = 0, heap = 294044
I (683) main: set blob @29, error = 0, heap = 294044
I (693) main: set blob @30, error = 0, heap = 294044
I (693) main: set blob @31, error = 0, heap = 294044
I (703) main: set blob @32, error = 0, heap = 294044
I (713) main: set blob @33, error = 0, heap = 294044
I (713) main: set blob @34, error = 0, heap = 294044
I (723) main: set blob @35, error = 0, heap = 294044
I (723) main: set blob @36, error = 0, heap = 294044
I (733) main: set blob @37, error = 0, heap = 294044
I (733) main: set blob @38, error = 0, heap = 294044
I (743) main: set blob @39, error = 0, heap = 294044
I (743) main: set blob @40, error = 0, heap = 294044
I (753) main: set blob @41, error = 0, heap = 293912
I (763) main: set blob @42, error = 0, heap = 293912
I (763) main: set blob @43, error = 0, heap = 293912
I (773) main: set blob @44, error = 0, heap = 293912
I (773) main: set blob @45, error = 0, heap = 293912
I (783) main: set blob @46, error = 0, heap = 293912
I (783) main: set blob @47, error = 0, heap = 293912
I (793) main: set blob @48, error = 0, heap = 293912
I (803) main: set blob @49, error = 0, heap = 293912
I (803) main: set blob @50, error = 0, heap = 293912
I (813) main: set blob @51, error = 0, heap = 293912
I (813) main: set blob @52, error = 0, heap = 293912
I (823) main: set blob @53, error = 0, heap = 293912
I (823) main: set blob @54, error = 0, heap = 293912
I (833) main: set blob @55, error = 0, heap = 293780
I (843) main: set blob @56, error = 0, heap = 293780
I (843) main: set blob @57, error = 0, heap = 293780
I (853) main: set blob @58, error = 0, heap = 293780
I (853) main: set blob @59, error = 0, heap = 293780
I (863) main: set blob @60, error = 0, heap = 293780
I (863) main: set blob @61, error = 0, heap = 293780
I (873) main: set blob @62, error = 0, heap = 293780
I (873) main: set blob @63, error = 0, heap = 293780
I (883) main: set blob @64, error = 0, heap = 293780
[... 515 lines cut ...]
I (3953) main: removed int @77, error = 0, heap = 294768
I (3963) main: removed int @78, error = 0, heap = 294768
I (3963) main: removed int @79, error = 0, heap = 294768
I (3973) main: removed int @80, error = 0, heap = 294768
I (3973) main: removed int @81, error = 0, heap = 294768
I (3983) main: removed int @82, error = 0, heap = 294768
I (3993) main: removed int @83, error = 0, heap = 294768
I (3993) main: removed int @84, error = 0, heap = 294768
I (4003) main: removed int @85, error = 0, heap = 294768
I (4003) main: removed int @86, error = 0, heap = 294768
I (4013) main: removed int @87, error = 0, heap = 294900
I (4023) main: removed int @88, error = 0, heap = 294900
I (4023) main: removed int @89, error = 0, heap = 294900
I (4033) main: removed int @90, error = 0, heap = 294900
I (4033) main: removed int @91, error = 0, heap = 294900
I (4043) main: removed int @92, error = 0, heap = 294900
I (4053) main: removed int @93, error = 0, heap = 294900
I (4053) main: removed int @94, error = 0, heap = 294900
I (4063) main: removed int @95, error = 0, heap = 294900
I (4063) main: removed int @96, error = 0, heap = 294900
I (4073) main: removed int @97, error = 0, heap = 294900
I (4083) main: removed int @98, error = 0, heap = 294900
I (4083) main: removed int @99, error = 0, heap = 294900
</pre>
</details>

tommybazar, 2018-02-20:
@agmuntianu 
I was just compiling this again to run it. Your logs illustrate the leak pretty well, but it looks like your NVS partition is kind of small, as blobs 70+ don't get saved at all.

My logs (with all blobs getting saved, but same thing in the end)

<details>
  <summary>Logs</summary>

(322) main: set blob @0, error = 0, heap = 271804
(324) main: set blob @1, error = 0, heap = 271620
(326) main: set blob @2, error = 0, heap = 271620
(330) main: set blob @3, error = 0, heap = 271620
(336) main: set blob @4, error = 0, heap = 271620
(340) main: set blob @5, error = 0, heap = 271620
(346) main: set blob @6, error = 0, heap = 271620
(352) main: set blob @7, error = 0, heap = 271620
(358) main: set blob @8, error = 0, heap = 271620
(364) main: set blob @9, error = 0, heap = 271620
(368) main: set blob @10, error = 0, heap = 271620
(374) main: set blob @11, error = 0, heap = 271620
(380) main: set blob @12, error = 0, heap = 271620
(386) main: set blob @13, error = 0, heap = 271620
(426) main: set blob @14, error = 0, heap = 271488
(428) main: set blob @15, error = 0, heap = 271488
(430) main: set blob @16, error = 0, heap = 271488
(434) main: set blob @17, error = 0, heap = 271488
(440) main: set blob @18, error = 0, heap = 271488
(446) main: set blob @19, error = 0, heap = 271488
(452) main: set blob @20, error = 0, heap = 271488
(456) main: set blob @21, error = 0, heap = 271488
(462) main: set blob @22, error = 0, heap = 271488
(468) main: set blob @23, error = 0, heap = 271488
(474) main: set blob @24, error = 0, heap = 271488
(480) main: set blob @25, error = 0, heap = 271488
(486) main: set blob @26, error = 0, heap = 271488
(490) main: set blob @27, error = 0, heap = 271488
(536) main: set blob @28, error = 0, heap = 271356
(538) main: set blob @29, error = 0, heap = 271356
(540) main: set blob @30, error = 0, heap = 271356
(544) main: set blob @31, error = 0, heap = 271356
(550) main: set blob @32, error = 0, heap = 271356
(556) main: set blob @33, error = 0, heap = 271356
(560) main: set blob @34, error = 0, heap = 271356
(566) main: set blob @35, error = 0, heap = 271356
(572) main: set blob @36, error = 0, heap = 271356
(578) main: set blob @37, error = 0, heap = 271356
(584) main: set blob @38, error = 0, heap = 271356
(588) main: set blob @39, error = 0, heap = 271356
(594) main: set blob @40, error = 0, heap = 271356
(600) main: set blob @41, error = 0, heap = 271356
(644) main: set blob @42, error = 0, heap = 271224
(648) main: set blob @43, error = 0, heap = 271224
(650) main: set blob @44, error = 0, heap = 271224
(652) main: set blob @45, error = 0, heap = 271224
(658) main: set blob @46, error = 0, heap = 271224
(664) main: set blob @47, error = 0, heap = 271224
(670) main: set blob @48, error = 0, heap = 271224
(676) main: set blob @49, error = 0, heap = 271224
(682) main: set blob @50, error = 0, heap = 271224
(686) main: set blob @51, error = 0, heap = 271224
(692) main: set blob @52, error = 0, heap = 271224
(698) main: set blob @53, error = 0, heap = 271224
(704) main: set blob @54, error = 0, heap = 271224
(710) main: set blob @55, error = 0, heap = 271224
(750) main: set blob @56, error = 0, heap = 271092
(754) main: set blob @57, error = 0, heap = 271092
(756) main: set blob @58, error = 0, heap = 271092
(758) main: set blob @59, error = 0, heap = 271092
(764) main: set blob @60, error = 0, heap = 271092
(770) main: set blob @61, error = 0, heap = 271092
[... 518 lines cut ...]
(5958) main: removed int @77, error = 0, heap = 270560
(5976) main: removed int @78, error = 0, heap = 270560
(5994) main: removed int @79, error = 0, heap = 270560
(6010) main: removed int @80, error = 0, heap = 270560
(6028) main: removed int @81, error = 0, heap = 270560
(6046) main: removed int @82, error = 0, heap = 270560
(6064) main: removed int @83, error = 0, heap = 270560
(6082) main: removed int @84, error = 0, heap = 270560
(6098) main: removed int @85, error = 0, heap = 270692
(6116) main: removed int @86, error = 0, heap = 270692
(6134) main: removed int @87, error = 0, heap = 270692
(6152) main: removed int @88, error = 0, heap = 270692
(6170) main: removed int @89, error = 0, heap = 270692
(6186) main: removed int @90, error = 0, heap = 270692
(6204) main: removed int @91, error = 0, heap = 270692
(6222) main: removed int @92, error = 0, heap = 270692
(6240) main: removed int @93, error = 0, heap = 270692
(6258) main: removed int @94, error = 0, heap = 270692
(6274) main: removed int @95, error = 0, heap = 270692
(6292) main: removed int @96, error = 0, heap = 270692
(6310) main: removed int @97, error = 0, heap = 270692
(6328) main: removed int @98, error = 0, heap = 270692
(6346) main: removed int @99, error = 0, heap = 270692

</details>

agmuntianu, 2018-02-20:
Yeah , I know about the size. I actually made the project from scratch and erased the flash, so this was allocated with whatever is the default parameters for the partition table.
The logs confirm the issue , it might be the root cause of my problem also .

negativekelvin, 2018-02-20:
It looks like blobs use 2 hashlist entries per blob as a new hashlistblock is created every 14 blobs vs 28 ints.  So probably hashlistbloocks are not being freed because both entries are not deleted since mHashList.erase is only called once or is called at the wrong time 
@igrr

igrr, 2018-02-20:
> It looks like blobs use 2 hashlist entries per blob as a new hashlistblock is created every 14 blobs vs 28 ints.

No, that's because HashListBlocks are not shared among pages, and 14 blobs of 255 bytes are enough to fill one page. So the HashListBlock still has some free entries, but the page is full. So we start writing to the next page, hence create a new HashListBlock.

The issue here is that HashListBlocks which have no items in them are deleted lazily, when the next HashListBlock::erase is called. So there's *at least one* HashListBlock per page hanging around, even if all the items in the page are deleted. I wouldn't go as far as calling this a memory leak (HashListBlock will be deleted along with the page if you re-initialize the storage), but i can understand why this looks unexpected. Fix is coming up.

negativekelvin, 2018-02-20:
@igrr ok that makes sense but why are none of the 9 blob HashListBlocks deleted as erase is being called while iterating the 100 items?

igrr, 2018-02-21:
There ends up being just one HashListBlock per page. As mentioned above, HashListBlock::erase removes blocks lazily — it first only removes hash item, and the next call can end up deleting the HashListBlock itself. The problem is that once there are no items left, the last HashListBlock does not get deleted. So I'm going to replace lazy deletion with deletion at once.

negativekelvin, 2018-02-21:
Something like this?

```C++
void HashList::erase(size_t index)
{
    for (auto it = mBlockList.begin(); it != mBlockList.end();) {
        bool haveEntries = false;
        bool foundIndex = false;
        for (size_t i = 0; i < it->mCount; ++i) {
            if (it->mNodes[i].mIndex == index) {
                it->mNodes[i].mIndex = 0xff;
                if (haveEntries)
                   return;
                foundIndex = true;
            }
            if (it->mNodes[i].mIndex != 0xff) {
                haveEntries = true;
            }
        }
        if (!haveEntries) {
            auto tmp = it;
            ++it;
            mBlockList.erase(tmp);
            delete static_cast<HashListBlock*>(tmp);
        } else {
            ++it;
        }
        if (foundIndex)
            return;
    }
    assert(false && "item should have been present in cache");
}

```

I still don't understand root cause of blob never finds and deletes empty blocks while int does.

negativekelvin, 2018-02-23:
Here is the result with the patch:
 <details><summary>Log</summary>

I (124) main: set blob @0, error = 0, heap = 292292
I (134) main: set blob @1, error = 0, heap = 292124
I (134) main: set blob @2, error = 0, heap = 291992
I (134) main: set blob @3, error = 0, heap = 291992
I (144) main: set blob @4, error = 0, heap = 291992
I (144) main: set blob @5, error = 0, heap = 291992
I (154) main: set blob @6, error = 0, heap = 291992
I (154) main: set blob @7, error = 0, heap = 291992
I (164) main: set blob @8, error = 0, heap = 291992
I (174) main: set blob @9, error = 0, heap = 291992
I (174) main: set blob @10, error = 0, heap = 291992
I (184) main: set blob @11, error = 0, heap = 291992
I (184) main: set blob @12, error = 0, heap = 291992
I (194) main: set blob @13, error = 0, heap = 291992
I (194) main: set blob @14, error = 0, heap = 291992
I (204) main: set blob @15, error = 0, heap = 291992
I (214) main: set blob @16, error = 0, heap = 291860
I (214) main: set blob @17, error = 0, heap = 291860
I (224) main: set blob @18, error = 0, heap = 291860
I (224) main: set blob @19, error = 0, heap = 291860
I (234) main: set blob @20, error = 0, heap = 291860
I (234) main: set blob @21, error = 0, heap = 291860
I (244) main: set blob @22, error = 0, heap = 291860
I (244) main: set blob @23, error = 0, heap = 291860
I (254) main: set blob @24, error = 0, heap = 291860
I (264) main: set blob @25, error = 0, heap = 291860
I (264) main: set blob @26, error = 0, heap = 291860
I (274) main: set blob @27, error = 0, heap = 291860
I (274) main: set blob @28, error = 0, heap = 291860
I (284) main: set blob @29, error = 0, heap = 291860
I (334) main: set blob @30, error = 0, heap = 291728
I (334) main: set blob @31, error = 0, heap = 291728
I (334) main: set blob @32, error = 0, heap = 291728
I (344) main: set blob @33, error = 0, heap = 291728
I (344) main: set blob @34, error = 0, heap = 291728
I (354) main: set blob @35, error = 0, heap = 291728
I (364) main: set blob @36, error = 0, heap = 291728
I (364) main: set blob @37, error = 0, heap = 291728
I (374) main: set blob @38, error = 0, heap = 291728
I (374) main: set blob @39, error = 0, heap = 291728
I (384) main: set blob @40, error = 0, heap = 291728
I (384) main: set blob @41, error = 0, heap = 291728
I (394) main: set blob @42, error = 0, heap = 291728
I (394) main: set blob @43, error = 0, heap = 291728
I (454) main: set blob @44, error = 0, heap = 291596
I (454) main: set blob @45, error = 0, heap = 291596
I (454) main: set blob @46, error = 0, heap = 291596
I (464) main: set blob @47, error = 0, heap = 291596
I (464) main: set blob @48, error = 0, heap = 291596
I (474) main: set blob @49, error = 0, heap = 291596
I (474) main: set blob @50, error = 0, heap = 291596
I (484) main: set blob @51, error = 0, heap = 291596
I (484) main: set blob @52, error = 0, heap = 291596
I (494) main: set blob @53, error = 0, heap = 291596
I (504) main: set blob @54, error = 0, heap = 291596
I (504) main: set blob @55, error = 0, heap = 291596
I (514) main: set blob @56, error = 0, heap = 291596
I (514) main: set blob @57, error = 0, heap = 291596
I (574) main: set blob @58, error = 0, heap = 291464
I (574) main: set blob @59, error = 0, heap = 291464
I (574) main: set blob @60, error = 0, heap = 291464
I (574) main: set blob @61, error = 0, heap = 291464
I (584) main: set blob @62, error = 0, heap = 291464
I (584) main: set blob @63, error = 0, heap = 291464
I (594) main: set blob @64, error = 0, heap = 291464
I (604) main: set blob @65, error = 0, heap = 291464
I (604) main: set blob @66, error = 0, heap = 291464
[... 524 lines cut ...]
I (3784) main: removed int @78, error = 0, heap = 293112
I (3794) main: removed int @79, error = 0, heap = 293112
I (3794) main: removed int @80, error = 0, heap = 293112
I (3804) main: removed int @81, error = 0, heap = 293112
I (3804) main: removed int @82, error = 0, heap = 293112
I (3814) main: removed int @83, error = 0, heap = 293112
I (3824) main: removed int @84, error = 0, heap = 293112
I (3824) main: removed int @85, error = 0, heap = 293112
I (3834) nvs_flash: Deleting empty HashListBlock 0
I (3834) main: removed int @86, error = 0, heap = 293244
I (3844) main: removed int @87, error = 0, heap = 293244
I (3844) main: removed int @88, error = 0, heap = 293244
I (3854) main: removed int @89, error = 0, heap = 293244
I (3864) main: removed int @90, error = 0, heap = 293244
I (3864) main: removed int @91, error = 0, heap = 293244
I (3874) main: removed int @92, error = 0, heap = 293244
I (3874) main: removed int @93, error = 0, heap = 293244
I (3884) main: removed int @94, error = 0, heap = 293244
I (3894) main: removed int @95, error = 0, heap = 293244
I (3894) main: removed int @96, error = 0, heap = 293244
I (3904) main: removed int @97, error = 0, heap = 293244
I (3904) main: removed int @98, error = 0, heap = 293244
I (3914) nvs_flash: Deleting empty HashListBlock 0
I (3914) main: removed int @99, error = 0, heap = 293380
</details>

negativekelvin, 2018-04-18:
The latest changes to nvs seem to improve this issue though the block deletion is not as quick as with the patch shown above. Maybe the performance trade-off is beneficial?

Oromis, 2019-02-19:
We ran into this issue as well (ESP-IDF v3.1.2). It causes our application to run out of memory over the course of a few days. This is intolerable for us and re-initializing NVS from time to time to get the memory back seems like an awful hack. 

I integrated @negativekelvin 's patch into a fork of `nvs_flash` and the leak disappeared completely. I don't see any negative effects of this patch either. It would be great to have this in master since updating a modified ESP-IDF version to new releases is a nightmare.

closed 2019-02-26 by igrr