espressif/esp-idf #1643: getting the a2dp_sink example to work

radiylonmina, 2018-02-19:
Hello, I've followed the steps to flash the example to my board and have configured the correct BT settings via 'make menuconfig', but I still can't manage to get the example to work.  Any ideas on what I should check/change or what direction I should start in debugging this?

ESP32DE, 2018-02-20:
Hi
need little more info from you
what does not work?
compile, flash, finding the ESP32 as BT Device bluetooth speaker, connect the BT Device, hear the Sound

so here a small list:
first part:
- does it compile? 
- you can upload it ( flash ) to the esp32 device?
- which esp32 modul you use?
- modul boots fine and wait for the BT connection? 

second part:
- do you use esp32 for sound output or do you use an DAC? Which?
- do you connect it by earphones / speaker?
- do you use a DAC that need I2C comands for loudness and so on?

hope this helps in the meantime to sort out your manage to get the example to work

best wishes
rudi ;-)

radiylonmina, 2018-02-21:
Hi Rudi,

Thanks for your help with this.  I'm able to compile and flash the example
to the ESP32 w/o any issues, but the board does not show up on BT capable
devices.  Also, when I attempt to make monitor,  I get a bunch of randomly
generated characters as a console output.

I plan on eventually connecting this to a speaker via an I2S codec, but for
now, I will use the onboard DAC to output low quality sound.

I am using the ESP32 Thing from the following site:
https://www.sparkfun.com/products/13907

Best,
Radiylon

pufstudio, 2018-02-23:
Afaik your devboard uses 26 Mhz crystal. You need to set correct settings in menuconfig.
Also, report if you get any sound using internal DAC. It used to work but I've been unable to make it work recently.

Weijian-Espressif, 2018-03-13:
@blueMoodBHD please help to view the issue, thanks.

blueMoodBHD, 2018-03-14:
Can you output the log now? if you can,please check whether BT  works well or not.
If "ESP_LOGI(BT_AV_TAG, "Audio packet count %u", m_pkt_cnt)" in bt_app_a2d_data_cb() have been printf, BT works well and the problem is in DAC settings.

closed 2018-03-26 by FayeY