espressif/esp-idf #1644: BLE Attribute Table/Service Table limits

dphilip7, 2018-02-20:
I have a set of custom services that I am trying to implements in a GATT server. To be specific, I have 8 services, 70 total characteristics, and around 60 total descriptors. I have 8 separate esp_gatts_attr_db_t objects, for each service, because trying to have more than one primary service in one threw an error. I guess I have two questions. One, is there a limit to how many service/characteristics/descriptors that can be created? In my debug log I am seeing that when my 7th service is created, I am getting an error. It seems to be due to checking if there is space in the attribute table to create the service. I thought it was due to the attribute handle number, but it seems as though you can have 0xFFFF handles, and I am no where close at ~222 (I'm not finished adding the last service and its characteristics yet). Here is part of the debug log.

```
I (457) GATTS_TABLE_DEMO: app_main init bluetooth

I (497) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 32

I (497) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 34

I (507) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 56

I (517) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 31

I (517) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 16

I (527) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 13

E (537) BT: GATTS_ReserveHandles: no free handle blocks

E (547) BT: service creation failed.
I (547) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 13

E (557) BT: service not created

I (557) GATTS_TABLE_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (567) GATTS_TABLE_DEMO: SERVICE_START_EVT, status 0, service_handle 72

I (577) GATTS_TABLE_DEMO: SERVICE_START_EVT, status 0, service_handle 106

I (587) GATTS_TABLE_DEMO: SERVICE_START_EVT, status 0, service_handle 162

I (587) GATTS_TABLE_DEMO: SERVICE_START_EVT, status 0, service_handle 193

I (597) GATTS_TABLE_DEMO: SERVICE_START_EVT, status 0, service_handle 209

I (607) GATTS_TABLE_DEMO: advertising start successfully
```

The part that is odd to me is that the attribute table creation says that it didn't fail (I am checking for the ESP_GATT_OK response), even thought the service wasn't created. The second question: is there a better way to set up these services that I am not following? I started using the table demo as a template since I would have multiple services with multiple characteristics each.

Any help is appreciated,

Dean

chegewara, 2018-02-20:
If you start from table demo example then in this line you have defined db table
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server_service_table/main/gatts_table_creat_demo.c#L174

HRS_IDX_NB is the number of handles that will be required and here is event that is called after db creation where you can check how many handles is assigned and if its the number equal to HRS_IDX_NB:

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_server_service_table/main/gatts_table_creat_demo.c#L453-L467

FayeY, 2018-02-23:
Hi @dphilip7 , did the above suggestion help you resolve the issue?

dphilip7, 2018-02-23:
Not quite, I don't have any problem adding additional characteristics to a service, it seems to be creating more than 6 attribute tables that is the problem.

I believe I have this part correct already. 
I made 8 enums for each service, which correspond to 8 different db tables.
I then use the esp_ble_gatts_create_attr_tab function to try and create the attribute table, and there is where I get the debug line "E (537) BT: GATTS_ReserveHandles: no free handle blocks"

It seems to not matter if I put all the characteristics from the extra services into existing services, but when I try to create a 7th attribute table for a 7th service I can never get that 7th service to work. I have tried multiple different service as the "7th service" so I don't think it's in the set up of the 7th service, although it's possible I'm missing something.

https://gist.github.com/dphilip7/05db6d7c501312f961f96de8071e5a04

chegewara, 2018-02-24:
There is for sure possible to create more than 6 services just im not sure about attributes table. With c++ wrapper im working with there was situation where user is been creating 20-ish services with many characteristics and if i remember 2 descriptors each characteristic.

dphilip7, 2018-02-24:
C++ wrapper? Is this open source, or are there any examples? I’m debating whether I should put in the time to try and create the services/characteristics/descriptors the long way without attribute tables or look for a different solution. I just liked the attribute table because it seemed like a simpler callback scheme.

chegewara, 2018-02-24:
https://github.com/nkolban/esp32-snippets

Weijian-Espressif, 2018-02-28:
@dphilip7 we limit service table, you can change the value of GATT_MAX_SR_PROFILES line 1022 in bt_target.h

dphilip7, 2018-02-28:
Perfect. That's exactly what I was looking for.
Is there any performance trade off of having more than the standard limit? I am planning on having BLE and Wifi running simultaneously, but not in a high performance setting. Mainly, I'm asking if, in creating more of these attribute tables, there is a ton of overhead on the processing or memory usage side?

Weijian-Espressif, 2018-03-05:
It's just for reducing memory

closed 2018-03-06 by dphilip7

lanzhihao, 2019-08-14:
how do you create multiple services in one profile. I tried with gatts_server_table_create demo but failed. First define 2 enums for services and use esp_ble_gatts_create_attr_tab() when ESP_GATTS_REG_EVT triggered. 
`enum
{
    IDX_SVC,
    IDX_CHAR_A,
    IDX_CHAR_VAL_A,
    IDX_CHAR_CFG_A,

    IDX_CHAR_B,
    IDX_CHAR_VAL_B,

    IDX_CHAR_C,
    IDX_CHAR_VAL_C,

    HRS_IDX_NB,
};

enum
{
    MY_SVC,
    MY_CHAR_A,
    MY_CHAR_VAL_A,

    MY_HRS_IDX_NB,
};
esp_err_t my_create_attr_ret = esp_ble_gatts_create_attr_tab(my_gatt_db, gatts_if, MY_HRS_IDX_NB, MY_SVC_INST_ID);`

lanzhihao, 2019-08-14:
How to fix that? I tried to add multiple services with gatt_server_table_create demo but failed. First define 2 enum for services and create attribute table when ESP_GATTS_REG_EVT triggered
`    enum
{
    IDX_SVC,
    IDX_CHAR_A,
    IDX_CHAR_VAL_A,
    IDX_CHAR_CFG_A,

    IDX_CHAR_B,
    IDX_CHAR_VAL_B,

    IDX_CHAR_C,
    IDX_CHAR_VAL_C,

    HRS_IDX_NB,
};

enum
{
    MY_SVC,
    MY_CHAR_A,
    MY_CHAR_VAL_A,

    MY_HRS_IDX_NB,
};
    esp_err_t my_create_attr_ret = esp_ble_gatts_create_attr_tab(my_gatt_db, gatts_if, 
    MY_HRS_IDX_NB, MY_SVC_INST_ID);`