espressif/esp-idf #1645: Failed Directed Advertising doesn't generate an LE Connection Complete

jakkra, 2018-02-21:
A HCI connection complete event is not received after performing a directed advertisement which fails to create a connection.

According to the [v5.0 core spec, page 1260](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=421043)

> If the Advertising_Type parameter is 0x01 (ADV_DIRECT_IND, high duty
cycle) and the directed advertising fails to create a connection, an LE
Connection Complete or LE Enhanced Connection Complete event shall be
generated with the Status code set to Advertising Timeout (0x3C).

HCI LOG:
```
CMD => 0A 20 01 00 // Disable advertise
EVT <= 0E 04 05 0A 20 00 //Success
CMD => 06 20 0F 40 06 D0 07 01 00 00 35 60 83 F3 12 00 07 00 // Set Adv Params, type 0x01
EVT <= 0E 04 05 06 20 00 // Success
CMD => 0A 20 01 01 // Enable advertise
EVT <= 0E 04 05 0A 20 00 // Success
We never get a LE Connection Complete here.
```

Weijian-Espressif, 2018-03-14:
@jakkra How to reproduce this issue?

jakkra, 2018-03-14:
Hi!

Start an advertisement of type `ADV_DIRECT_IND, high duty cycle`. According to the spec a LE Connection Complete HCI event shall be generated if no device connects to us within some timeout, with Status code 0x3C. 

We aren't using the Bluedroid stack, but this problem should affect it too, since this is in the HCI layer. I don't have an example application for you, because I think it should be easy enough to reproduce.

I guess using the bluedroid stack, all you need to do is to start directed advertising with high duty cycle and then watch the HCI log, correct behaviour is a LE connection complete with status code 0x3c.

Weijian-Espressif, 2018-03-15:
@jakkra thanks for your help , Our controller will generate a LE connection complete with status code 0x3c.  you can add log in btm_ble_conn_complete() of btm_ble.c in line 1995.

jakkra, 2018-03-15:
@Weijian-Espressif Thanks!

closed 2018-03-19 by FayeY