espressif/esp-idf #1646: Task watchdog got triggered - it is fixed with a vTaskDelay of 10ms but is this a bug?

shaoner, 2018-02-21:
Hi,

I have a ESP32-DEVKIT-C running, I wanted to create a simple task:

```C
void io_task(void *parameter) 
{
    while(1) {
        vTaskDelay(2 / portTICK_PERIOD_MS);
    }
}

void app_main(void)
{
    nvs_flash_init();
    xTaskCreate(io_task,"io_task",2048,NULL,1,NULL);
}
```
Here I got this warning:

> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 1)
> Tasks currently running:
> CPU 0: IDLE
> CPU 1: io_task
> 

So I tried different strategies like using `esp_task_wdt_reset()` in the while, but it fails and returns:
`
ESP_ERR_NOT_FOUND: Error, the current running task has not subscribed to the TWDT
`

The only way I can make it work is by using:
```C
   while(1) {
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
```
But I have no idea why?

projectgus, 2018-02-21:
Hi @shaoner,

The default tick rate is 100Hz (you can [configure this](http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#freertos-hz) but 100 is a good tradeoff between ISR overhead and responsiveness).

This means portTICK_PERIOD_MS is 10 (10ms per tick). Therefore a 2ms vTaskDelay is a 0 tick delay, ie no delay at all, which is interpreted as "yield to higher priority tasks, otherwise keep running". Therefore `io_task` will starve lower priority tasks (like IDLE) if it never blocks waiting for something else to happen.

You can [disable the task watchdog timer](http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#task-wdt) if you're sure that you want this task running on the CPU as much as possible, and don't mind starving lower priority tasks. Bear in mind this may have unexpected consequences (the idle task is used for cleanup of deleted tasks, and the FreeRTOS timer task also runs at low priority).

Alternatively you could set the tick rate to 1000Hz. FreeRTOS will spend more time in context switch interrupts, but you'll be able to delay a task for 2ms.

closed 2018-02-21 by projectgus

akshar001, 2018-08-17:
Anyone who still want to avoid that error should try
adding these three line to your continuous  rtos function, 

`TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;`
 `TIMERG0.wdt_feed=1;`
 `TIMERG0.wdt_wprotect=0;`

And add this header
```
#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"
```

moefear85, 2021-09-05:
The Answer in my opinion is made of 2 parts. The first has been answered by @projectgus (see above). The 2nd part of the solution is:

From what I've come to understand... ESP-IDF creats a watchdog timer for idle tasks in addition to any tasks doing your work. IE, it creates two extra tasks IDLE0 & IDLE 1 (one for each core), whose sole purpose is to do nothing, ie idle. These idle tasks simply feed the watchdog whenever its respective core is idling. This means by design, esp-idf requires and forces that each core never be fully busy for too long. Whenever this happens (such as an infinite loop that doesn't call vTaskDelay or its equivalent anywhere), then it triggers the watchdog, because that core is now fully busy, meaning it is no longer idling often enough, meaning the idle "tasks" can't run ie reset the watchdog ie effectively verify that the core has infact idled recently enough.. IE you are defacto not allowed to fully consume the CPU and MUST have every task pause somewhere, to give the IDLE tasks some time to "run". The framework ensures this by giving them the lowest priority. IE, the reason is that the IDLE tasks are starving for "execution time", is because they have by default the lowest possible priority that the framework allows, so that any other task created will always have a higher priority, where if any task having a higher priority never takes a break (such as by calling vTaskDelay), then no task of lower priority (including the IDLE tasks), can "run".

In other words there is no nice solution. Either disable the watchdog fully, or modify the framework to remove the IDLE tasks or somehow give them higher priorities (equal or greater than your task) while being careful however that they then don't block your work tasks (if they can do that), or simply don't allow any task to fully consume a core, so that it idles frequently enough for the watchdog to stay happy. Another alternative would be increasing the time to trigger the watchdog. I don't know how to do this, and I don't know what the max limit is.

Also: I'm not sure if the task, when it performs a blocking call (such as a blocking IO read that takes a long time) results in idling thus gives the IDLE tasks time to run, or whether it counts as "execution" thus further starving the IDLE tasks and triggering the watchdog. In the former case, you wouldn't need to add any delays, unless the calls end up not blocking long enough (then you'll have to artificially idle by calling vTaskDelay). In the latter it means adding a delay wouldn't help, you'll have to set time limits on the blocking call to force it to return before the watchdog triggers, and even then, unless you do something else that effectively idles, you'll still have to call vTaskDelay or equivalent.

monopol40, 2021-10-30:
Problem still exists. I tried all ideas above given without success. No stable load a file to SD_MMC via websockets  where message is comprised of multiple frames or the frame is split into multiple packets throws exception, not always but often:
Example in monitor:
ws[/ws][1] frame[0] binary[10044 - 10411]: E (75417) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (75417) task_wdt:  - async_tcp (CPU 0/1)
E (75417) task_wdt: Tasks currently running:
E (75417) task_wdt: CPU 0: IDLE0
E (75417) task_wdt: CPU 1: async_tcp
E (75417) task_wdt: Aborting.

Dazza0, 2022-08-24:
@monopol40 This sounds like an actual TWDT timeout (i.e., the `async_top` task is actually consuming too much CPU time). Could you please double check that `async_top` task doesn't hog the CPU, and has adequate delays while it's running (or is event driven).